ControlSystemStudio/phoebus #1373: Question: training material

georgweiss, 2020-06-04:
At ESS users show a lot of creativity when creating OPIs. This is a problem when features like scripting and rules are being used to an extent that borders to abuse.

My question to the community is if there exists any material (documentation, training) addressing aspects like best practices and tips-n-tricks with Phoebus. I'm assuming here that not all users are veterans with long experience with EDM and BOY, so I think there is a use case for this type of material that could be shared by all sites.

kasemir, 2020-06-04:
Slides from training sessions (https://controlssoftware.sns.ornl.gov/training/2019_USPAS/, https://controlssoftware.sns.ornl.gov/training/2019_USPAS/Presentations/05%20DisplayBuilder.pdf, going back to BOY, https://controlssoftware.sns.ornl.gov/training/2011_CEA/2_2_BOY.ppt) kept emphasizing to keep it simple:

1) Add a widget
2) Enter PV name
3) Done

Don't try to create scripts when you have no idea how to do that.
Don't turn a Label into a text update, don't turn a circle into an LED, don't turn a group of polygons into a button, and then complain in 3 years that the next tool doesn't render that to the same pixel.

https://github.com/ControlSystemStudio/phoebus/wiki/Display-Builder-Compatibility is a growing collection of notes on how to create displays that increase your likelihood of being compatible with the next tool.

georgweiss, 2020-06-04:
Thanks @kasemir . However, my issue is that users actually do know how to create scripts. And that they think programming is more fun than spending time to figure out how to actually solve a particular problem using the built-in features in Phoebus.

georgweiss, 2020-06-04:
In any case, your links contain material I think should be consulted by every OPI developer. Would it be OK to clone one or several of the slide decks and adopt to Phoebus?

kasemir, 2020-06-05:
Yes, https://controlssoftware.sns.ornl.gov/training/ already states

_Feel free to use the material offered here for your local EPICS introductions. When modifying the material for local use, please include a reference to the original material, for example in the form of a "Based on https://controlssoftware.sns.ornl.gov/training/" link._

closed 2020-12-19 by georgweiss