ControlSystemStudio/phoebus #1380: Release Guildeline and Procedure

shroffk, 2020-06-05:
Discussion:

It is great that over the last few months we have been able to have multiple releases of phoebus.
The release process has been quick and simple...this should help us push out changes and bug fixes quickly to our users and improve the general user experience.

With phoebus moving into production at multiple sites (SNS and ESS have had it in the control room for awhile now) I think this would be a good time to formalize and document the release procedure. The practices at each site and the good practices associated with putting out a release that would keep everyone happy.

Initial thoughts.
Release:

1. pre release:
 - [ ] Notification or PR (the release plugin can be configured to push to a branch which we can merge via a PR with approval... the NSLS2 product, run-scripts, and ansible based deployment process  needs to be [updated](https://github.com/shroffk/nsls2-phoebus/commit/547f88efc21c07e7ff1c6219370ac04016a9c252) with new releases, so such a notification would be helpful)

2. release process:
 - [ ] Releases should be tagged. (the release plugin does this already)
 - [ ] Source and Common product binaries hosted on the github project. (instead of working hard on automating this we could simply as a community upload win, mac, and linux platform tars)
 - [ ] Change log and Release Notes
 - [ ] prepare for next release, update version etc. (we will have to co-ordinate and find a solution on how the pom and build.xml are both updated)
 - [ ] tech-talk announcement?

ralphlange, 2020-06-05:
Is your argumentation only about the maintenance/bugfix digit? I would generally agree.
Or are you arguing against using numbered versions at all? I would strongly disagree.

ralphlange, 2020-06-05:
fwiw: ITER has strict requirements on traceability. That's why for "external" products a released version is preferred over a git hash - under the assumption that a released version has a release note, complete and matching documentation etc etc.
A date is not unique and does not provide an acceptable level of traceability.

shroffk, 2020-06-05:
I think we could add our use cases and requirements first. It might be better to sort out the details once they are all collected.

At NSLS2

I have 2 different products for the accl and beamline.
A ansible based deployment solution which build the latest phoebus product and the alarm services on the target machines.
Currently, we checkout the master and build that. In the accelerator control room there are talks about wanting a more rigidly controlled version/tag/release.

kasemir, 2020-06-05:
Hi Ralph, great Friday discussion, right?

I'm not specifically arguing for a certain numbering scheme, but trying to point to the issue:

I need the update from some PR 123. When I look at an installed product, I need to be able to tell: Does this include the update from PR 123, or is it old, and I need to replace it?

The current "release" number doesn't tell me that.

ESS incremented the "release" number for their needs, so they can tell by the release number if the product includes what they care about or not. But those release numbers which are useful to ESS don't help me at all. Quite the opposite: I needed to accordingly increment numbers in my deployment scripts.

So I guess everybody cares about specific but different PRs. An update that's important to you doesn't matter to me and vice versa. Should we only increment "release" numbers for you? No. Do we increment them for every PR? Well, then we might as well use the date.

kasemir, 2020-06-05:
One solution for your accl vs beamline could be: Beamlines use the date/time as version information. For accl, you add tags "ITER_ACCL_1.2.3".

I have the same, but different time line, so again SNS beam lines use the date/time as version, and I add tags "SNS_ACCL_3.4.5" to identify the more rigidly controlled updates.

I think something like that would be more useful than one site deciding what the release is for everybody, or everybody trying to agree on a common release cycle.

ralphlange, 2020-06-05:
Hash is good for traceability, but not sorted, so you can't tell what is newer.

kasemir, 2020-06-05:
Sorry, when I say "date" I mean "date + time", and to avoid time zone confusion we can use the UTC time. It's basically the hash, but easier to compare. "Anything after June 1" is easier for a human to check than "anything after d789a7dcbefb316b1c697394f3b9f050eb87fdc4"

ralphlange, 2020-06-05:
Major and minor numbers - per convention - indicate different kinds of compatibility.
Losing that, and having no idea which update breaks compatibility and which doesn't, would be a disaster.
Also, supporting multiple release series is hard if you can't tell them apart.

ralphlange, 2020-06-05:
Date and time is *not* the hash, btw. Not at all.

ralphlange, 2020-06-05:
Again, for a specific release series, e.g. 4.5.x, 'date-time-hash' would work for x, as long as they are increasing and uniquely identify a repository commit.
Across release series and compatibility breaks - not really.

kasemir, 2020-06-05:
Right, maintaining several branches in parallel requires a lot of effort.
I don't see that.
At best, we have enough people to maintain the 'master' of the Eclipse-based legacy and the new CS-Studio.

ralphlange, 2020-06-05:
How would you mark compatibility breaks?

kasemir, 2020-06-05:
Well, Kunal changed the context menu API in #1375. From that PR on, it's different. Previous site-specific code needs to change. If somebody wasn't ready to follow, they'd have to move onto a branch with the old API.

If the organization was larger, you could have a committee to plan the API change, prepare the branch, and keep maintaining that branch for a certain time.

georgweiss, 2020-06-06:
I'd like to better understand the apparently negative impact on the deployment procedure reported by @kasemir . Can you please provide details?

kasemir, 2020-06-07:
See link in Kunal's introductory post, https://github.com/shroffk/nsls2-phoebus/commit/547f88efc21c07e7ff1c6219370ac04016a9c252.