ControlSystemStudio/phoebus #1387: Formula PV: formula PV appears disconnected though its input PVs are connected

pavel-ch, 2020-06-10:
We are extensively using formula PVs with 2 and more input PVs. Irregulary happens, that input PVs are connected but Fomula PV itself is disconnected (=has a value with DISCONNECTED alarm) after display startup. Proven by Probe, too.
My guess it that this is caused by initialization of PVs on display startup together with a threading issue:
The method `org.phoebus.pv.formula.FormulaInput.handleUpdate(VType)`:

```
    private void handleUpdate(final VType value)
    {
        logger.log(Level.FINE, () -> formula_pv.getName() + " updated by " + pv);
        variable.setValue(value);
        formula_pv.update();
    }

```
is called from more threads. If interrupted between `setValue `and `update ` calls, it can happen that the old initially disconnected input PV value wins over the new connected value.
I tried to make this method `synchronized` and the effect has gone.
However I do not know further consequences of making this method synchronized.
Can you please evaluate and let me know if the change is OK @kasemir ?
Thank you.

pavel-ch, 2020-06-10:
Well it appears that
> ... and the effect has gone.

is not really true. Effect is still there.
So please take this Issue as an misbehavior notification.
I must dig a bit deeper to find the real reason.
At the moment seems that setting PV logging level from WARNING to FINE really switches the effect off, perhaps because changing thread timing on my system.

pavel-ch, 2020-06-11:
I tested your code, the issue is not observed so far. Needs a longer run. I would propose to merge to master.

kasemir, 2020-06-11:
Thanks for testing. Just found one more thing: It's possible that the evaluation on thread X used an older value from an input updated on thread Y because of an omitted `volatile`.

pavel-ch, 2020-06-11:
Right this could be the originating issue. For me, problem did not show since the decoupling introduction.

kasemir, 2020-06-11:
Both are certainly useful improvements. Have to see if the latter was in fact the original problem.

Anything special in how you get into that state?
Is it formulas with many inputs? Or did you see it with just 2 inputs? I assume never with just 1 input?
I assume these inputs are using your MQTT-type PVs? Or do you see it with plain local and simulated PVs?
Does it only happen with the "initial" state after opening a display? When later changing one of the inputs, does the formula then properly update? Or does it also happen with a long-running formula that keeps updating, but for some reason it remains disconnected even though its inputs are changing?

pavel-ch, 2020-06-11:
I can test the old code with the volatile keyword, do you think it helps?
To your questions:
- the issue appeared irregularly (=sometimes and for different PVs) and relatively rare. We use many fPVs with 2 inputs, much less with 3 and more. No with 1 input as far as I know. Issue observed on 2 input fPVs but cannot say it was limited to them.
- yes inputs are my MQTT "hmi://" pv type. No usage or tests on other PV types.
- yes it only happen with the "initial" state after opening a display. And yes, always when later changing one of the inputs, the formula properly updates and "connects". Formulas are short mostly of type ='PV1' + 2 * 'PV2'

kasemir, 2020-06-11:
Sure, testing the original code where you saw the issue, but now with `volatile`, would be useful.

Good to know that it's only about the "initial" state, though not sure, yet, what that means.
I bet logging changes matters, plus it tends to result in a flood of messages, but anyway:
If the log level for  org.phoebus.pv is FINE, then FormulaInput.handleUpdate logs when it's invoked. The question would then be: For those formulas that are initially 'disconnected', was handleUpdate called for all its inputs with a non-null value? Or are we for some reason missing one of the initial values?

pavel-ch, 2020-06-11:
As written above, switching PV logging level to FINE effectively masked the issue away somehow, so I was not able to see any one issue appearance with FINE level and inspect the real update order. Will do the test, but thinking about the above, the "volatile" may not matter. Will see.

pavel-ch, 2020-06-11:
Well, proven, even with volatile the issue shows on old code. So what really helped is the decoupling.

krisztianloki, 2020-07-29:
Hmm... I have the opposite problem: the referenced PV is disconnected but the formula PV is not.

georgweiss, 2020-07-30:
I'm looking at @krisztianloki 's issue and it's obviously linked to the initial state of a formula PV. On the other hand, if the widget launches with all input PVs connected it will behave as expected if any of the PVs disconnects later.

krisztianloki, 2020-08-06:
#1442 fixed the problem with formulas like ```=2*`no_such_pv` ```  but not for ```=`no_such_pv`?0:1```. The latter still shows up as connected.
Moreover, if the referenced PV disconnects during runtime the widget shows the usual INVALID/DISCONNECTED decoration but if the referenced PV was initially disconnected then it only shows INVALID/UDF decoration

georgweiss, 2020-08-06:
Please refer to issue and PR for information on why this was implemented this way.

kasemir, 2020-08-06:
You're correct on
```
=`no_such_pv`?0:1
```

I assume a bug in the formula implementation of ` .. ? .. : ..`. I'll look at it.

kasemir, 2020-08-06:
The validity of the `condition` in `=condition ? ok : else` was indeed not checked.
Now these all look correct:

[image]

pavel-ch, 2020-11-18:
After some months I return to this issue:
> We are extensively using formula PVs with 2 and more input PVs. Irregulary happens, that input PVs are connected but Fomula PV itself is disconnected (=has a value with DISCONNECTED alarm) after display startup. Proven by Probe, too.

As we are already in production now, I am forced to solve also this issue as it is really showing.
Using a set of log messages, I think I was able to find out the main reason.
My communication PV data source (hmiPv) reacts sometimes very fast with the first update of a newly created PV. On a wide set of our PVs, this first update is **the only one** for a long time (e.g. power switch state on a power distribution substation). This first update comes from another thread (MQTT Rx thread) than the thread the PV has been created on.
It seems, that for Formula PV, this (fast) first update of underlying PV(s) is sometimes lost due to a threading issue.

My view of the reason:
When constructing `FormulaPV`, the `FormulaInput `is instantiated for each variable. Subsequently, the PV is created from `PVPool `(`FormulaInput:41`) and `handleUpdate` is asked to subscribe on `pv.onValueEvent`. This ends up in `PV.java:86 addSubscription`. Inside this function, it sometimes happens, that the thread context switches in between `final VType value = last_value;` and `value_subs.add(value_sub);`. The first value update comes from MQTT thread but because the PV is not yet subscribed (not added to `value_subs` list), the update is lost and the old initial DISCONNECTED `value` is set to Formula PV.

This happens rarely, but happens. More often on a heavy loaded system.

Unfortunately, I do not see a way how to do PV subscription "atomic" without locking.

What can help is connecting the communication PV to the external source not from its constructor, but for example after the first subscription. This can ensure having the first live PV update AFTER having the subscription reliably in effect. For that, I would need an access to the `addSubscription` method and `ValueEventHandler.Subscription` class from PV derived types.

@kasemir Kay could you please consider the case? This could perhaps affect other PV types, too, if the first update comes from another thread that way. 

Thank you.

pavel-ch, 2020-11-18:
Okay, if I do locking in PV,java:
`private Object value_lock = new Object();`
```
    void addSubscription(final ValueEventHandler.Subscription value_sub)
    {
        synchronized (value_lock) {
            // If there is a known value, perform initial update
            final VType value = last_value;
            if (value != null) {
                logger.log(Level.FINE, name + " value updated from addSubscription: " + value);
                value_sub.update(value);
            }
            value_subs.add(value_sub);
            logger.log(Level.FINE, name + " addSubscription: subscription added, value: " + last_value);
        }
    }
```
and:
```
    protected void notifyListenersOfValue(final VType value)
    {
        synchronized (value_lock) {
            last_value = value;
            for (ValueEventHandler.Subscription sub : value_subs)
            {
                try
                {
                    sub.update(value);
                }
                catch (Throwable ex)
                {
                    logger.log(Level.WARNING, name + " value update error", ex);
                }
            }
        }
    }
```
All is working fine.
But I guess the locking is not really desired here due to performance penalties?

kasemir, 2020-11-18:
Need to think about it. Generally best to not hold any locks when invoking callbacks/event handlers. Holding a lock while invoking `sub.update(value)` can sooner or later deadlock, thinking about user code that would try to get just one value, so it un-subscribes inside that 'update' implementation, or creates a new PV and then subscribes to that one inside 'update'  etc.

kasemir, 2020-11-18:
> The first value update comes from MQTT thread through notifyListenersOfValue,
> but because the PV is not yet subscribed (not added to value_subs list),
> the update is lost and the old initial DISCONNECTED value is set to Formula PV.

Since you seem to have a setup where that happens often enough so you can test it, does it help to simply move  `value_subs.add(value_sub);` to the start?
```
void addSubscription(final ValueEventHandler.Subscription value_sub)
{
    value_subs.add(value_sub);

    // If there is a known value, perform initial update
    final VType value = last_value;
    if (value != null)
        value_sub.update(value);
}
```

pavel-ch, 2020-11-18:
I did some tests and have not observed the effect. So the situation improved significantly. What I see is that there is still a risk of context switch in between
```
    final VType value = last_value;
    if (value != null)
        value_sub.update(value);

```
This could potentially update subscriber with old value.
But seems that the risk is significantly lower than before.

kasemir, 2020-11-18:
First, thanks so much for identifying the issue!
While I briefly considered looking at the formula PV code, I think you're correct that this needs to be fixed at the PV level, where as you describe there's a problem with first checking for an already received value, then adding to the subscriber list. It can indeed drop the initial (and maybe only) update in case that happens in between the check and the list addition.

Swapping the code by first adding to the subscriber list, then checking if there's already an initial value to send out adds two new issues.

If the first (and maybe only) update arrives in between, we would notify on that first value twice. Once in the 'normal' notifyListenersOfValue chain, plus via the initial update in addSubscription.

Alternatively, there could be a race between the 'initial' update an another 'normal' update. Running in parallel, the 'initial' could appear later, and  an outdated value thus replaces the current one.

Need to think about it for a while, if this really requires locking, or some additional flags.

pavel-ch, 2020-11-18:
From my perspective, notifying the same update twice is not a real problem. Updating in reverse order with the old value last would be a real pain, but in my fast test I have not seen that, indicating a lower probability of this event.

kasemir, 2020-11-18:
Agree that double updates are tolerable, certainly for the formula.
But ending with the old value is real bad.

Simply locking would fix it, but adds the potential for deadlock.
Locking and moving all notifications to another thread so they're outside the lock, would fix everything, i.e. basically this:

```
private SomeConcurrentQueue<VType> notification_queue;

synchronized void addSubscription(final ValueEventHandler.Subscription value_sub)
{
        value_subs.add(value_sub);
        if (last_value != null)
             notification_queue.push(last_value);
}

synchronized void notifyListenersOfValue(final VType value)
{
    last_value = value;
    notification_queue.push(last_value);
}

// Some new notification thread:
void run()
{
    while (true)
    {
        VType value =  notification_queue.blocking_pop();
        for (ValueEventHandler.Subscription sub : value_subs)
            sub.update(value);
    }
}
```

pavel-ch, 2020-11-18:
For my usage this looks good. As this is a really core functionality I would generally be cautious about performance and a potential risk of growing queue size in case of fast value updates, slow update processing...

kasemir, 2020-11-18:
I have the same concerns.
Still, the current implementation where it's occasionally dropping the one and only value is clearly sub-par.
Moving `value_subs.add(` up helps, but we can end up with an old value replacing the latest.
Plain synchronize can deadlock.
While the likelihood of these is getting smaller compared to the original issue, they're all still pretty bad.

I'll profile to determine how fixing all of it adds to the CPU load.

kasemir, 2020-11-18:
Locking, then avoiding deadlocks by decoupling the events seemed to nicely handle the original issue.
But it breaks the `ArrayPVDispatcher` as used by the array widget and likely several other unit tests that depend on `loc://..` PVs reacting right away to a 'write'...

Your original suggestion to `synchronize` works much better!
Will look into possible deadlocks, and maybe use `Semaphore.tryAcquire` so that it can warn and abort instead of deadlocking.

pavel-ch, 2020-11-28:
Looks like the issue has been solved. Closing.

closed 2020-11-28 by pavel-ch