ControlSystemStudio/phoebus #1389: Thumbwheel, Knob, Clock, Gauge Widget

Amadrath, 2020-06-11:
The widgets mentioned above seem to be disabled, when I start phoebus on my Kubuntu 20.04. After trying to resolve every warning, poping up during phoebus startup, and making sure the latest versions of javaFX are available to phoebus (Focal fossa comes with openjfx 11), nothing worked. I found the hidden_widget property, and emptied the list. Then some of the widgets were available in the Display Builder, but were not working. Am I correct in assuming, that these widgets are not fully developed, yet?

georgweiss, 2020-06-11:
You are correct, these widgets are not fully implemented. However, the thumbwheel widget is under development and will be merged in the near future.

Amadrath, 2020-06-11:
It might be a good idea, to add a corresponding remark in the examples window, to avoid the appearance of some problem with the individual instance of phoebus.

georgweiss, 2020-06-12:
I'm sorry, not quite sure what you mean. The widgets you mention are not mentioned in the example displays. Can you please clarify?

Amadrath, 2020-06-12:
When you open File -> Top Resources -> Example Display, a window with Buttons for widget examples opens. The Action Buttons for the mentioned widgets disappear/are disabled and an according message appears in the console.

shroffk, 2020-06-15:
@georgweiss & Frederik, there is interest in the pinwheel widget from NSLS2 and fhi-berlin (there was a request from their users today). 
Is it possible for the widget to be contributed in a independent module for the time being. We can test and debug it and then merge when everyone is comfortable.

shroffk, 2020-06-15:
> You are correct, these widgets are not fully implemented. However, the thumbwheel widget is under development and will be merged in the near future.

@georgweiss & Frederik, there is interest in the pinwheel widget from NSLS2 and fhi-berlin (there was a request from their users today). 
Is it possible for the widget to be contributed in a independent module for the time being. We can test and debug it and then merge when everyone is comfortable.

kingspride, 2020-08-07:
are there any news on this? asking for a friend. I mean for my users at the FHI ;)

shroffk, 2020-08-07:
Yes...at yesterday's monthly meeting it was agreed that a new named branch with these widgets will be created for testing
@georgweiss

kingspride, 2020-08-07:
thanks for the info. looking forward to it !

kingspride, 2020-09-07:
sorry for bothering but I'd like to get another update on this topic, if there is already progress?

shroffk, 2020-09-08:
@georgweiss ping for fredrik :)

fredrik-soderberg, 2020-09-09:
I am working on the Thumbwheel and LinearMeter widgets. At the previous monthly meeting we agreed on to include the ThumbWheel widget on a new named branch as a first step. I am currently testing the thumbwheel widget and hope to be able to share the code soon. After this I will focus on the linear meter. thanks!

kingspride, 2020-11-20:
leaving again a ping on this - how is the progress?