ControlSystemStudio/phoebus #1401: duplicate declaration in core-framework

jeonghanlee, 2020-06-25:
@kasemir and @shroffk 

You don't like this warning again, but I would like to know, somehow from the fresh compling without the exist .m2 directory, I have to remove L29 of the two duplicated dependency 

* https://github.com/ControlSystemStudio/phoebus/blob/master/core/framework/pom.xml#L12
* https://github.com/ControlSystemStudio/phoebus/blob/master/core/framework/pom.xml#L29

```
Downloading from central: https://repo.maven.apache.org/maven2/org/springframework/boot/spring-boot-dependencies/2.1.5.RELEASE/spring-boot-dependencies-2.1.5.RELEASE.pom
[ERROR] [ERROR] Some problems were encountered while processing the POMs:
[WARNING] 'dependencies.dependency.(groupId:artifactId:type:classifier)' must be unique: org.phoebus:core-formula:jar -> duplicate declaration of version 4.6.4-SNAPSHOT @ org.phoebus:core-framework:[unknown-version], /home/jhlee/gitsrc/phoebus-env/phoebus-src/core/framework/pom.xml, line 27, column 17
[ERROR] Non-resolvable import POM: Could not transfer artifact org.springframework.boot:spring-boot-dependencies:pom:2.1.5.RELEASE from/to sonatype-nexus-snapshots (https://oss.sonatype.org/content/repositories/snapshots): proxy: Name or service not known @ org.phoebus:service-alarm-logger:[unknown-version], /home/jhlee/gitsrc/phoebus-env/phoebus-src/services/alarm-logger/pom.xml, line 17, column 19
[ERROR] 'dependencies.dependency.version' for org.springframework.boot:spring-boot-starter:jar is missing. @ org.phoebus:service-alarm-logger:[unknown-version], /home/jhlee/gitsrc/phoebus-env/phoebus-src/services/alarm-logger/pom.xml, line 29, column 17

```

Please let me know what you think. 

@kasemir , sorry about the maven building dependency issues.

shroffk, 2020-06-25:
@jeonghanlee good catch,
we can remove the duplicate definition

kasemir, 2020-06-25:
Do you need to keep the one with or the one without `<scope>compile`?

shroffk, 2020-06-25:
By default no scope should mean compile...
Looking back this duplicate dependency was declared cause I was having problem getting eclipse to recognize the dependency.
I think I have a previous issue ticket about java error in eclipse when importing phoebus. I have a PR which removes the duplicate

shroffk, 2020-06-25:
@jeonghanlee this should be resolved now. Thanks for checking out the warnings and helping clean up :)

closed 2020-06-25 by jeonghanlee