ControlSystemStudio/phoebus #1403: Gdk-WARNING

libaf, 2020-06-26:
Hi,
I get the following error when I start up Phoebus and click Applications/Display/New Display:
"(java:20070): Gdk-WARNING **: 16.52:58.167: XSetErrorHandler() called with a GDK error trap pushed. Don't do that."
How can I resolve that?
CentOS 7, java-11-openjdk
Cheers,
Markus

kasemir, 2020-06-26:
Do you always get this warning?
Is it just the warning, otherwise everything works OK?
Do you still get the warning when you run with `-Djdk.gtk.version=2`?

libaf, 2020-06-27:
Yes, the warning seems to appear whenever the Display window is open (haven't noticed it otherwise yet, but I've just started trying out Phoebus). From what I can tell, everything works as it should otherwise.

Tried running with ./phoebus.sh -Djdk.gtk.version=2 but it "ignores the launch parameter":
2020-06-27 12:09:02 WARNING [org.phoebus.ui.application.PhoebusApplication] Ignoring launch parameter '-Djdk.gtk.version=2'

kasemir, 2020-06-27:
`-Djdk.gtk.version=2` is not a launch parameter for the phoebes app but for the JVM.
In phoebus.sh, you need to add it right after 'java':

```
java -Djdk.gtk.version=2 -jar $JAR $OPT "$@" &
```

libaf, 2020-06-29:
Sorry for the misunderstanding. With that parameter added to the shell script I don't get the warning. Thanks! Should I update some library dependency in order for this to work without that statement?

kasemir, 2020-06-29:
https://bugs.openjdk.java.net/browse/JDK-8156779

kasemir, 2022-03-16:
The warning is still happening with Openjfx18, #2179, but doesn't seem to have any negative effect