ControlSystemStudio/phoebus #1409: MySQL Version

kasemir, 2020-07-01:
Dependabot #1408 suggests updating the MySQL client lib from 5.1.45 to 8.0.16.

That's easy enough to do as in commit 9c8fe2f6f95425712f43aec2860583426a1b39c6, but there are compatibility issues.
The MySQL client 5.1.45 works fine with a very old MySQL 5.1.73 server as well as a slightly newer  MariaDB 5.5.65.
The new  MySQL client 8.0.16 won't connect to them, giving a "CLIENT_PLUGIN_AUTH is required" error.

A brief web search suggests that these server/client versions are simply incompatible.
The operational setups to which I need to interface require the older client, so I need to stay with the currently used MySQL client version.

If you don't use MySQL at all, no worries.

Once the first user comes up with a need to interface a new MySQL server and the old client lib will fail to connect, we can reconsider.

kasemir, 2022-09-13:
No change:

https://github.com/ControlSystemStudio/phoebus/security/dependabot/2
https://github.com/ControlSystemStudio/phoebus/security/dependabot/20
https://github.com/ControlSystemStudio/phoebus/security/dependabot/21

georgweiss, 2022-09-14:
I have not verified it, but in order to muffle Dependabot we could maybe treat the MySQL jdbc driver in the same manner as the Oracle jdbc lib?

kasemir, 2022-09-15:
Our oldest MySQL setup went away, and the current mysql-connector-java 8.0.30 works fine with another one I tested, so I'll just create a `${mysql.version}` property in the parent pom, set that to the current version 8.0.30, and dependabot will be happy.
That should work for most people, and in case a local setup needs to fall back to an earlier version, they can patch the version in the parent pom before building the site-specific binaries.

closed 2022-09-15 by kasemir