ControlSystemStudio/phoebus #1410: opi "menu" widget with no label (i.e. label is an empty string) is translated to .bob "ActionButton" widget with label "No PV"

oksanagit, 2020-07-01:
Hi,
Please take a look at the attached pic. 
I actually made a fix for it and will create a PR in a few. 

[image]

kasemir, 2020-07-02:
How do you get that display?
Are you converting from MEDM to *.bob?
Or are you opening a *.opi display (which had previously been converted from MEDM)?

oksanagit, 2020-07-02:
I am opening an .opi display, which was previously converted from MEDM.  We would like to move to the new product as smooth as possible, and  it is my understanding that same code is being used in both scenarios: opening an .opi screen and translating from .opi to .bob and opening bob. Thank you.

kasemir, 2020-07-02:
There are 2 routes:

1) MEDM -> *.opi (using the converter in BOY) -> open *.opi in Display Builder
2) MEDM -> *.bob (using the converter in the display builder)

You chose 1), which is fine.
Can you attach the *.opi file?

oksanagit, 2020-07-02:
[pointGrey.opi.txt](https://github.com/ControlSystemStudio/phoebus/files/4864790/pointGrey.opi.txt)
I had to add .txt. .opi extension was not accepted as legitimate.

kasemir, 2020-07-02:
Thanks.
Can you add ADPlugins.opi, since that's the one with the actual menu buttons in question:

[image]

oksanagit, 2020-07-02:
[ADPlugins.opi.txt](https://github.com/ControlSystemStudio/phoebus/files/4865580/ADPlugins.opi.txt)
Yes, sorry, this is an included display, the main part of the story.

oksanagit, 2020-07-02:
Let me know it anything else is needed. Thanks

kasemir, 2020-07-02:
OK thanks, will look at it.

kasemir, 2020-07-02:
Try with the latest. Thanks for finding this.

oksanagit, 2020-07-06:
I have tested the master branch, looks good to me. Thank you Kay!

closed 2020-07-06 by oksanagit