ControlSystemStudio/phoebus #1420: Databrowser - passing / using macros in databrowser

pavel-ch, 2020-07-15:
1) I was not able to pass macros to databrowser using standard DB action syntax:
[image]
What I get (on Windows) is always an exception:
[image]
Because URI was treated as a filename including macros.
I would suggest, in `org.csstudio.trends.databrowser3.DataBrowserInstance.loadResource(URI)`, use:
`XMLPersistence.load(new_model, ResourceParser.getContent(new URI(input.getScheme() + "://" + input.getPath())));`
instead of original:
`XMLPersistence.load(new_model, ResourceParser.getContent(input));`
This works for me. Can you approve @kasemir ?

2) In databrowser, using macro for PV name generally works, but on several places shows macro name instead of expected resolved text like here:
[image]

3) I would really appreciate to have a possibility to macroize start / end time. This is suitable for loading one PLT file with many time ranges. I can do it, can you approve @kasemir ?

Thanks

kasemir, 2020-07-15:
Sounds good, thanks

kasemir, 2020-07-30:
I think this has been handled, re-open if not.

closed 2020-07-30 by kasemir