ControlSystemStudio/phoebus #1431: Ordering of widgets in xml impacts rendering

georgweiss, 2020-07-28:
When the attached opi file is opened in edit mode the TextUpdate and SliderButton widgets are not rendered, screen shot attached. Here is what I have observed:
1) If the PV name of the Label widget is shortened such that the entire text fits within the width, then all widgets are rendered correctly.
2) If the TextUpdate or SliderButton widgets are selected in the outline, all widgets are rendered correctly.
3) If the opi file is edited manually to put the Label widget last, all widgets are rendered correctly even if the PV name of the Label widget is wider that the width if the text field.

The obvious conclusion is that the rendering order - defined by the order in the opi xml file - causes this behavior. Have not been able to determine whether this is an JavaFX or Pheobus issue.

I am able to reproduce this issue consistently in edit mode. In runtime the problem comes and goes in a seemingly unpredictable manner.
In any case, the ordering in xml should of course not matter.
[image]

[en-dis-chan.bob.zip](https://github.com/ControlSystemStudio/phoebus/files/4988376/en-dis-chan.bob.zip)

kasemir, 2020-07-28:
> the ordering in xml should of course not matter.

The ordering matters very much. It determines what's "on top of" what, and also sets the "tab" order when navigating between widgets. 
The order of widgets in the XML can be set in the editor via the context menu (Move to back, ...) and in the outline/tree view.

georgweiss, 2020-07-28:
I do not understand, the widgets are sized and positioned to not overlap.

kasemir, 2020-07-28:
The order still matters for "Tab" order.

In any case, your problem can be fixed by reverting the slide button to 'managed'.
In SlideButtonRepresentation line ~155:
```
        // This code manages layout,
        // because otherwise for example border changes would trigger
        // expensive Node.notifyParentOfBoundsChange() recursing up the scene graph
        hbox.setManaged(false);
```

Change that to 'true' or just remove since that's the default, and it properly renders.

georgweiss, 2020-07-28:
OK, I will change as it seems to solve the issue.

kasemir, 2020-07-28:
Might also have to check the resize. On the TextEntryRepresentation you can see how it needs to use 'setPrefSize' vs. 'resize' based on managed or not.

georgweiss, 2020-07-28:
How does that work with autosize?

kasemir, 2020-07-28:
Exactly, please check that the size updates correctly when autosize is enabled.

closed 2020-07-31 by georgweiss