ControlSystemStudio/phoebus #1435: Spaces in paths&names

DorianHas, 2020-07-29:
Hello there,

We encounter an issue with Phoebus recent version. Spaces in names and directories aren't took into account. It's creating the following error message : 

[image]

Dorian

kasemir, 2020-07-29:
Long story short: You best avoid spaces as well as characters like `(`, `)`, `:` in file names, or come up with a perfect rewrite of everything that works for everybody.

In the end, the resource names must be URIs. We need to support access to the same `display.bob` or `plot.plt` no matter if it's a local `file://path/to/plot.plt`, which is typically the case in a control room, or if it's accessed as `http://my.site.org/display_files/path/to/plot.plt`, which is often the case when users run the tool in their office. Inside a URL, spaces need to be escaped as `%20`. The technical difficulty is that when we write files, like from a display editor or the data browser, that's of course only possible with local files, not via URIs, which are read-only. So we do need to do a certain amount of conversion between URI and local files. Our code has undergone several simplifications to move from string manipulations to basic Java URI/Path/.. API, but it's tricky because for example Windows uses different file name rules than Mac OS and Linux.

kasemir, 2020-07-29:
The problem was introduced in #1422, and I think I've found a work around that'll again allow spaces in names.

kasemir, 2020-07-30:
Fixed

closed 2020-07-30 by kasemir