ControlSystemStudio/phoebus #1438: Something wrong with editing Label or widget's PV name

haitao127, 2020-07-31:
When using the latest version(master branch) of phoebus, I can't edit the content of Labels or PV name of widgets directly by double click the widget and then single click it, if I do so, it will return back to uneditable state.

kasemir, 2020-07-31:
Could this be a side effect of #1049?

pavel-ch, 2020-07-31:
Should not, looking into the code changes in #1049, I think that closing the inline editor after losing focus is the same like before. Seems that we get the focus = false into the listener.

Workaround: Use keyboard only for inline editor edits, move caret with arrows and do not mouse click there.

kasemir, 2020-07-31:
Sure, keyboard works.
Still, after the double-click to open an inline editor, the content (PV name or label's text) is fully selected.
Is is somewhat intuitive to then click one more time with the intent to position the cursor.
I haven't checked why we then actually 'commit' and close the editor, and frankly don't have time to do that right now.

pavel-ch, 2020-07-31:
I fully agree, the first thing one makes after the doubleclick is one more click to get rid of selection.
I try to look at it but not right now.

pavel-ch, 2020-07-31:
Seems the reason is in:
`org.csstudio.display.builder.editor.EditorGUI.createElements()` line 398:
`layout.addEventFilter(MouseEvent.MOUSE_PRESSED, event -> layout.requestFocus());`
After a mouse click, layout gets focus and inline editor loses it and closes.
Introduced on 21.5.2020 by by Kay at #1340 with 68261f5 Editor: Get keyboard focus on mouse press

kasemir, 2020-07-31:
Good find! Yes, the problem is that the keyboard focus doesn't always follow the mouse.
You try to copy/paste/delete a widget, but instead it does that to some text in a property or vice versa.
So clicking there requests the focus.
Maybe keep the basic idea but check if there's an active inline editor?

kasemir, 2020-07-31:
YES:
```
  layout.addEventFilter(MouseEvent.MOUSE_PRESSED, event ->
        {
            if (! this.editor.getSelectedWidgetUITracker().isInlineEditorActive())
                layout.requestFocus();
        });
```

kasemir, 2020-07-31:
I'll do a PR. Thanks so much for figuring this out!

closed 2020-07-31 by kasemir