ControlSystemStudio/phoebus #1446: Inconsistent widget disable state

georgweiss, 2020-08-03:
Widgets associated with a disconnected PV are rendered somewhat inconsistently in runtime. In the attached screen shot:
1) CheckBox, SlideButton and BooleanButton are disabled.
2) Scaled slider is disabled, but alarm border is not.
3) LED and TextUpdate are enabled.

There may be additional widgets to consider.

In any case: what is the expected runtime behavior?

[image]

kasemir, 2020-08-03:
The only guaranteed runtime behavior is that you get a border which indicates that the widget has a problem.

The text entry and text update widgets which usually show the PV's value will instead show the `<PV name>`.

As for enabled/disabled, originally all widgets were 'disabled' when not connected by disabling the underlying JavaFX widget. But that would, well, disable the widget, so it no longer reacts to the normal button presses, context menu etc. That has the severe downside that you can no longer open the context menu and inspect which of the one or more PVs associated with the widget are disconnected. Which is a bummer, because the first thing you need to do is usually to access the context menu and copy the PV name, or view archived data to see when the problem started etc. So instead of disabling the JavaFX widget, a `not_enabled` style is applied which  tends to visually resemble the disabled state (adds opacity), but the widget still reacts to some extend.
Details depend on the widget. For some, the `not_enabled` style is applied (e.g. BoolButton). With others, some part of the widget is disabled, but others stay active to support the context menu (e.g. ActionButton). It's a balance between minimizing the Scene graph size (just use one JavaFX widget and apply a style) and getting the desired functionality (action button has a top-level Pane to host the actual Button or MenuButton that can change at runtime).

georgweiss, 2020-08-03:
In my view a consistent UI behavior is imperative as lack of it confuses users and triggers creation of support tickets. I'd like to propose that all widgets show the same (or as similar as possible) behavior with respect to tool tips, context menu and disabled state when PV is disconnected.

georgweiss, 2020-08-04:
This is in my view unanswered. User is asking for a consistent disable state, so I am prepared to go ahead implementing it.

georgweiss, 2020-08-07:
I hope this can be discussed and resolved: what is the initial state of a widget associated with a disconnected PV? It should not matter whether the widget is a monitor (read only) or control (read/write).

kasemir, 2020-08-07:
A disconnected widget indicates via its border that it's not connected.
A 'control' widget will in addition appear disabled.

georgweiss, 2020-08-18:
I'd like to fix the Spinner widget to consistently handle a disabled PV. However, there is the question on how to support the context menu. As it stands currently the "Phoebus context menu" is shown only when right-clicking the up or down button, a right click on the text field part of the widget triggers the default select/copy/paste menu. At the same time a right click in the TextEntry widget input field does not trigger the default select/copy/paste menu but instead shows the Phoebus context menu.

Suggestion is to let the input field in the Spinner widget also trigger the Phoebus context menu as it is not obvious that only a right-click on the spinner buttons can trigger it.

georgweiss, 2020-08-28:
@kasemir , @shroffk and others, can we decide on this? The spinner widget needs a fix, but we need also to decide if the JavaFX native copy/paste menu should be supported or not. See previous comment.

closed 2020-09-02 by georgweiss