ControlSystemStudio/phoebus #1448: File browser monitors resources outside of current file browser path?

kasemir, 2020-08-03:
When a file browser is set to `/some/top/level/path`, for example based on the memento persisted from a previous shutdown, it opens up with that path and displays the files in there, as expected.

At least on Mac OS, however, it also seems to monitor other folders, specifically the home directory, resulting in these types of messages after running for some time:
```
WARNING [org.phoebus.applications.filebrowser] Cannot refresh /Users/fred/.bash_sessions within /some/top/level/path
WARNING [org.phoebus.applications.filebrowser] Cannot refresh /Users/fred/.bash_history within /some/top/level/path
```

Unclear if this is a bug in the file browser not properly watching/un-watching folders, or if it is a perculiarity of the Mac OS file system.

krisztianloki, 2020-08-03:
I have similar log entries under Linux:
```WARNING [org.phoebus.applications.filebrowser] Cannot check for /home/krisztianloki/.viminfo.tmp within /home/krisztianloki/Development/ess-opis/NON-APPROVED
WARNING [org.phoebus.applications.filebrowser] Cannot refresh /home/krisztianloki/.viminfo.tmp within /home/krisztianloki/Development/ess-opis/NON-APPROVED
WARNING [org.phoebus.applications.filebrowser] Cannot refresh /home/krisztianloki/.viminfo.tmp within /home/krisztianloki/Development/ess-opis/NON-APPROVED
WARNING [org.phoebus.applications.filebrowser] Cannot check for /home/krisztianloki/.viminfo within /home/krisztianloki/Development/ess-opis/NON-APPROVED
```

kasemir, 2020-08-03:
Thanks for checking. So it's likely a bug in the code that watches/ignores folders as the path is changed. Not a huge issue, but unnecessary. I'll look at it.

closed 2020-08-03 by kasemir