ControlSystemStudio/phoebus #1452: Scroll issues seen on Debian

tanviash, 2020-08-04:
Seeing this issue on Debian 10 OS
•	Sometimes scrolling with scroll wheel doesn't work when on top of certain elements like linking containers. Some containers can be scrolled, so it makes sense for them to get scroll events first, but if they cannot be scrolled any further, they should pass the event to the page level, and scroll the page.
•	Similarly, ctrl-scroll wheel does not zoom in/out when mouse is over linking containers.

kasemir, 2020-08-04:
From https://github.com/kasemir/org.csstudio.display.builder/issues/557:

Yes. The up/down scrollbar works, you can also use the mouse wheel to scroll up/down when the mouse is in most of the display, but the mouse wheel cannot scroll up/down when your mouse pointer is on top of the embedded display because of ControlSystemStudio/phoebus#1363 and ControlSystemStudio/phoebus#1355.
When the embedded content is larger so that the embedded display itself has a scroll bar, then the mouse wheel will scroll within that embedded display.

cogansco, 2020-08-04:
In the most frustrating cases, our overview screens, most of the page consists of embedded content, which means the scroll wheel doesn't work most of the time.  In these cases, scrolling the embedded view is disabled or not necessary, due to correct sizing, however there is still the need to scroll the main page.  CS-Studio handles this correctly.  Can it be addressed in Phoebus?

kasemir, 2020-11-17:
I'd say the focus handling of the scroll wheel has been fixed.
When opening "Structure..Embedded" from the example displays and making the window small enough to get scroll bars, the wheel will scroll the overall page except when the mouse is on top of the small embedded screen which itself has scroll bars, in which case those 'inner' display is scrolled.