ControlSystemStudio/phoebus #1461: Reading from waveform PV only returns the first element

krisztianloki, 2020-08-07:
Using a waveform PV as a string and trying to display it in a Text Update will only show the first character. It is the same with the probe tool; changing the format does not help. It works with 4.6.3 though

kasemir, 2020-08-07:
What's the data type? Must be a CHAR waveform.

kasemir, 2020-08-07:
Example: PV is a char waveform, plain probe shows the characters, but with Format: String you get the chars as a string.

[image]

krisztianloki, 2020-08-07:
It is a CHAR. I was able read it in the Eclipse based CS-Studio and Phoebus-4.6.3 but not with the development version.
But when I tried again just now I can... I don't quite understand... I'll try to reproduce it next week...

krisztianloki, 2020-08-07:
So this is what I did:
 - set the waveform to `foobar` using the latest Phoebus
 - the Text Entry changed from `foobar` to just `f`
 - confirmed that the string was correctly set using both `caget` and `dbgf` in the IOC
 - stopped latest Phoebus and started 4.6.3
 - Phoebus-4.6.3 correctly displayed `foobar`
 - stopped 4.6.3 and started latest Phoebus
 - latest Phoebus now displays `foobar`

kasemir, 2020-08-07:
> set the waveform to foobar using the latest Phoebus
> the Text Entry changed from foobar to just f

Cannot reproduce. Entering new text works OK with a waveform like this:
```
record(waveform, "text")
{
  field(FTVL, "CHAR")
  field(NELM, "200")
}
```

[image]

krisztianloki, 2020-08-07:
It seems to work with a local IOC... maybe it has something to do with the CA gateway? Reloading the OPI also fixes the problem.

kasemir, 2020-08-07:
There might be a problem with the gateway not handling changing array sizes.
We also never use a gateway with write access, ours are all read-only, so cannot check.

krisztianloki, 2020-08-07:
You are right. It was the gateway. `camonitor` also showed the first character only and the array size did not change. Thanks for the help!

closed 2020-08-07 by krisztianloki

ralphlange, 2020-08-07:
Please file an issue with the Gateway. It handles changing array sizes on monitor connections, and should handle this case.
https://github.com/epics-extensions/ca-gateway/issues