ControlSystemStudio/phoebus #1463: XY graph autoscaling

shroffk, 2020-08-12:
We are trying to use the xygraph for both horizontal and vertical plots.

I have 2 waveform/array pv's. One pv holds a list of indices while the other holds the values

When I use the indices for the x axis and the value for the y axis then the autoscaling of both works fine.

When I reverse this the autoscaling of the y axis does not work... I am not sure I understand what is going wrong here. 
Sometimes simply adding a few extra elements to either of the arrays trigger the scale to be calculated correctly.

I added breakpoints to the rtplot and verified that it has all the correct data points (all 9 of them)...with the correct x and y value.

[image]

shroffk, 2020-08-12:
here is the example screen:

[xf-grpah.zip](https://github.com/ControlSystemStudio/phoebus/files/5063055/xf-grpah.zip)

kasemir, 2020-08-12:
In PlotProcessor.autoscale, the value range is only checked for samples within the current X range.
Maybe print out the result of `determinePositionRange` and see if that explains what's happening.

shroffk, 2020-08-12:
Thanks, 
You are kind of right, the main issue is in the following
```
// Last sample is the one just inside end of range.
int stop = search.findSampleLessOrEqual(data, position_range.getHigh());
```
in the determineValueRange

The way the stop index is determined is the cause of the above behaviour.

The position range is being calculated correctly.... in the above case it is 0-100. The stop index is calculated at "4", since that is a index where the y value is 100.

kasemir, 2020-08-12:
For tests you can replace that start..stop determination with
```
start = 0;
stop = data.size()-1;
```

I guess we need to skip that start..stop determination and just use 0.. data.size()-1  if the 'position' data is not ordered (because in this case it's basically swapping the 'position' and 'value').

shroffk, 2020-08-12:
I feel like I am clearly missing some assumption about the values being used as index...cause moving some values results in the complete range being used 

[image]

kasemir, 2020-08-12:
The find-sample stuff is a binary search. For ordered data, that's the fastest.
For non-ordered data, moving one value can get different results.

shroffk, 2020-08-12:
> For non-ordered data, moving one value can get different results.

yes,
I was confused if it was using the value, but I see it is doing a search using the X value which the assumption it is ordered.

> For tests you can replace that start..stop determination with
>
> ```
> start = 0;
> stop = data.size()-1;
> ```
>
> I guess we need to skip that start..stop determination and just use 0.. data.size()-1 if the [...]

just saw this,
replacing the start/stop determination works.

but... Would this break other things? Should I just not be doing this?
This is a very common use case used in almost all our camera screens.

kasemir, 2020-08-12:
> Would this break other things?

With a "normal" data set, ordered X waveform, zooming "in" means we only auto-scale the visible data.
With start=0, stop=size-1 we would always auto-scale based on the full value range, not just the visible value range.

I think the best is to bite the bullet and check if the position data is ordered. A little expensive, ORD(N), but if it is ordered, you then still get some benefit from the binary search and the reduced number of samples to check when zooming in.

I'll look at doing that soon.

kasemir, 2020-08-12:
With the update in the PR, your example now works, and since it checks for ordered or not, autoscale still behaves when zooming in on the X axis in "normal" plots with ordered X data.

closed 2020-08-12 by shroffk

oksanagit, 2020-08-18:
Hi Kay and Kunal,
The fix does not work for me. I have reproduced the problem with ADSimDetector, please see attached screen. I will be happy to share these .bob files (main screen needs embedded displays to run) for testing. Please LMK.  This is with Int16, Stats5 enabled, Intensity graph shows PVA  Image. Thank you. 

[image]

kasemir, 2020-08-18:
What's the data?
Can you update Kunal's example with that data?
Replace the "loc://xxx(1, 2, 3, 4, ..)" arrays with one of your value snapshots.

oksanagit, 2020-08-18:
Please see attached. I included Kunal's graph with updated values, a few. and a text file with caget PV_name redirected to it. These are real profile values. 
Thank you,
[xyPlot.zip](https://github.com/ControlSystemStudio/phoebus/files/5092989/xyPlot.zip)
[image]

kasemir, 2020-08-19:
I think the issue is that Kunal's original example had both 'X' and 'Y' waveforms, your example only has an 'X' waveform with 'Y' auto-created based on the index. Will look at it.

kasemir, 2020-08-19:
I was wrong, having just an 'X' waveform was not the problem.
Issue was in the X axis autoscale when the data wasn't ordered.
Your example works now.