ControlSystemStudio/phoebus #1468: Paste properties does not update the Properties view

krisztianloki, 2020-08-14:
When using the copy-paste properties function the Properties view is not updated. Have to select a new widget and go back to see the changes and the display needs to be reloaded for the changes to take effect.

kasemir, 2020-08-17:
Cannot duplicate.
Tried for example:
1) Create Label and Rectangle.
2) De-select 'transparent' on the label
2) Select Rect, copy properties 'background color'
3) Select Label, paste properties -> Label background updates, as does the color indicated in the properties.
4) Create another label
5) Select first label, edit text, copy properties 'text'
6) Select second label, paste properties -> Label text updates, as does the text shown in the properties view

krisztianloki, 2020-08-18:
Then it will be a bug in the `Text Symbol` widget; copying the `Symbols` property produced the result I described. The `Symbol` widget even threw an exception.

krisztianloki, 2020-08-24:
So, it seems that neither `ArrayWidgetProperty` nor `StructuredWidgetProperty` calls `firePropertyChange()` in `readFromXML()`.
In `ArrayWidgetProperty.readFromXML()` it is possible to call `setValue(elements)` instead of `value = elements` and it solves the issue but `StructuredWidgetProperty` overrides `setValue()` and throws an `IllegalAccessError` so `firePropertyChange()` needs to be explicitly called.
However, I'm not sure if there are side effects of adding these new notifications...

krisztianloki, 2020-08-27:
Fixed by #1497

closed 2020-08-27 by krisztianloki