ControlSystemStudio/phoebus #1487: Macros

libaf, 2020-08-24:
Hi,
I have a number of macros defined in my startup-script for an EPICS IOC. I would prefer if the macros were accessible in Phoebus in order to increase the modularity. Is there a way to do that?
Cheers,
Markus

kasemir, 2020-08-24:
As your system grows, your IOC might load more than one database.
A common use for macros is loading the same database file with different settings, somewhat like this:
```
dbLoadDatabase("motors.db", "CARD=1,MOT=1")
dbLoadDatabase("motors.db", "CARD=1,MOT=2")
dbLoadDatabase("motors.db", "CARD=1,MOT=3")
...
dbLoadDatabase("motors.db", "CARD=2,MOT=1")
dbLoadDatabase("motors.db", "CARD=2,MOT=2")
dbLoadDatabase("motors.db", "CARD=2,MOT=3")
...
```

When you say "I would prefer if the macros were accessible in Phoebus" I wonder which value you would assign to `$(CARD)` and `$(MOT)` in the above example. Eventually, you might even have more than one IOC!

Be that as it may,  as soon as you figure out what macros you want to make available in phoebus, you can read the online help to learn how to define them, see 

https://control-system-studio.readthedocs.io/en/latest/index.html
https://control-system-studio.readthedocs.io/en/latest/preferences.html
https://control-system-studio.readthedocs.io/en/latest/preference_properties.html#display-builder-model

For example, you may create a file `my_settings.ini`:
```
# My settings
#
# Pass on the phoebus command line via
#    -settings /path/to/my_settings.ini

org.csstudio.display.builder.model/macros=<CARD>1</CARD><MOT>2</MOT>
```

libaf, 2020-08-24:
We will be using ethercat plc hardware modules and their .db files will be loaded multiple times.  The naming convention of these modules in the IOC follows that of ESS, for example IOC:Master-SlaveNbr-EL2602-BO1. If I were to insert/exchange a module at the very start of the plc modules, that would mean that I would potentially need to adjust the linking of all the slaves after that module in Phoebus. For reasons of modularity I would therefore want to link a certain slave number to a macro that, for example, could be named after the valve it controls.

Thanks a lot, that was exactly what I was looking for as a start! If I am to use this approach, I might need to make the startup-script for the IOC export the macros of interest into a file and then load these into the settings.ini-file of Phoebus. Is there another way of doing this, could I for example load such a file of  macro-definitions into Phoebus at startup (e.g. maybe using the -resource argument)?

ralphlange, 2020-08-24:
I would always argue that implementation information (in your case the address - slave number - and the record type - BO) should _never_ be made part of the record name.
If you change your implementation (e.g. by inserting another module at the beginning or by using an mbbo record because you need a third value option) you are in big trouble if implementation is part of the name. Changing names in an existing system is a very very big and dangerous procedure. (Just try to imagine how the archive retrieval would handle this...)

It's much better to keep such internal/address information in a bunch of separate stringin records - so that any client can easily find out the hardware or implementation related bits.
So - in your case: put the valve name into the record name and the slave number into a record VALVE:SlaveNbr. Your names won't change no matter how many slaves you add and every client (not just Phoebus) can find out the mapping at any time.

georgweiss, 2020-08-24:
Sorry for the intrusion, but deployment at ESS is a factor here. Please contact me on georg.weiss@ess.eu.

kasemir, 2020-08-24:
You shouldn't have to load any macro definitions during phoebus startup.

Assume you have two motors.
You would _NOT_ start one instance of phoebus with CARD=1, MOT=1 to talk to the first motor, then exit it, and start a different instance with CARD=1, MOT=2 to talk to the second motor.

Instead, you have a top-level "motors.bob" with for example embedded displays for each motor (where the embedded display widget sets CARD, MOT as necessary), or you have buttons to open per-motor displays (where the button actions to "open display" set the macros).

For an introduction to the display builder in general which also mentions macros see https://controlssoftware.sns.ornl.gov/training/2019_USPAS/

ralphlange, 2020-08-24:
Generally speaking:
If you have a strong need for e.g. an additional "technical" view (where things have names based on their address, byte and bit numbers, etc), record name aliases could be an option.
Using aliases opens several cans of worms, though, and must be carefully planned and implemented.

libaf, 2020-08-25:
@ralphlange @georgweiss : Just to clarify, LIBAF is a smaller lab which do not have the manpower to implement a full standalone solution. We are therefore thankful for the opportunity to use a lot of the work of ESS. Still, thanks for the suggestion regarding the naming of records.

@kasemir Sure, of course that makes sense for the case you present. I think we might be talking past each other though? The reason I would like to import a file of macro definitions from the startup-script of the IOC into Phoebus is the naming convention which our lab uses (described above). So, in order to increase the potential for modularity, I would like to use a macro named $(Z-V1)=SlaveNumber for the first valve of our Zero-degree beamline.  With this I would be able to call the record in Phoebus with for example *****-s$(Z-V1)-BO1. If, in the future,  I would put a hardware module before  “Z-V1”, the number of the slave will update itself without me having to change the code in Phoebus.

georgweiss, 2020-08-25:
OK, I see. I was under the impression that you worked for ESS.

libaf, 2020-08-27:
Thanks for your replies.

Unknown to me before, ecmc includes the option to add an alias to a record via the startup-script. This solves the problem for us.

Cheers

closed 2022-03-03 by kasemir