ControlSystemStudio/phoebus #1490: Travis build for trusty and openjdk11 failing

shroffk, 2020-08-24:
Currently, 2 PRs are failing while creating the travis environment for running the CI

Found some mention in travis issues which claimed the following...

```
It looks like only Trusty is affected:

dist: trusty: https://travis-ci.org/cotsog/travis_ci_prod_regular/builds/525229081 5
dist: xenial: https://travis-ci.org/cotsog/travis_ci_prod_regular/builds/525229251 7
Is it an option for you to add dist: xenial to your config? Probably a good move given Trusty will be EOL at the end of the month 4.

Let me know what you think!
```

Going to try to updated travis to use xenial

ralphlange, 2020-08-24:
Just as a general remark: Keep in mind that Trusty is as close as we can get to RHEL 7 - which many sites use.
This is probably affecting C/C++ code more than Java, though.

shroffk, 2020-08-24:
`"export DISPLAY=:99.0; sh -e /etc/init.d/xvfb start;" failed and exited with 127 during`

With the update to xenial we need to use the new service as described below

https://docs.travis-ci.com/user/gui-and-headless-browsers/#using-xvfb-to-run-tests-that-require-a-gui

closed 2020-08-24 by shroffk