ControlSystemStudio/phoebus #1493: Alarm datasource

shroffk, 2020-08-25:
The old cs-studio had a datasource which allowed accessing information from various parts of the alarm tree. 

Phoebus would benefit from having this same type of alarm source linked to the new phoebus alarm server.

ralphlange, 2020-08-25:
Pretty please!
+1

shroffk, 2020-08-31:
Started with an "alarm" datasource.
Currently, you can create alarmpvs or the root, or any node, or the leaf.

The pv is of type VString
The un acknowledged alarms are mapped to the VType equivalent, so these pv's can directly applied to widgets and the alarm borders should work.

shroffk, 2020-08-31:
I decided to not go the VTable route as the old datasouce since I don't think it is very useful form a visualization perspective.
sticking to scalar types would lend itself far better to mapping this datasource to existing widgets...

Next steps will be to spec out the write operations and the subchannels (like value, alarm time, ack/unack, etc...)

shroffk, 2020-08-31:
support is needed for entries into the alarm server of pvs with datasource
```
/NSLS2_OPR/SR/Vac/ca://pv1
/NSLS2_OPR/SR/Vac/pva://pv1
```

closed 2020-11-25 by shroffk