ControlSystemStudio/phoebus #1508: Ways to check the status of localizations

tmijail, 2020-08-31:
On PR #1495, @pavel-ch pointed out that silently falling back to the english localization if a message wasn't present on `messages_xx.properties` would make maintaining a localization a pain because there would be no easy way to check whether there were messages missing from `messages_xx.properties`. 
He would previously watch the log for `{CLASS} is missing value for '{MESSAGE}'`, which now only appears if `{MESSAGE}` is missing from both `messages.properties` and `messages_xx.properties`.

I found two ways to make this easy again.

### 1. Localization Tools Maven Plugin
We generate a [report ](https://web.archive.org/web/20090105015431/http://mojo.codehaus.org/l10n-maven-plugin/report-mojo.html) that looks something like this.  

[image]
([live report from another project](https://maven.apache.org/plugins/maven-pmd-plugin/l10n-status.html))

We'd just need to add [this profile](https://github.com/tmijail/phoebus/blob/7a4ab7c8c87230e9fbf262f217298016956bf4da/pom.xml#L207) to the pom, and then it's as easy as running `mvn site -P l10n-status`. The report will be located at `phoebus/target/site/l10n-status.html`.

### 2. Checkstyle
We can use Checkstyle's [translation](https://checkstyle.sourceforge.io/config_misc.html#Translation) rule and one of the [available plugins](https://checkstyle.sourceforge.io/#Related_Tools) to integrate Checkstyle with the IDE (or just use the Checkstyle CLI).
I've tried the [IDEA plugin](https://github.com/jshiell/checkstyle-idea) and it's pretty nice. 

[image]

You can double-click on the warnings to take you to the correct file and then use IDEA's Resource Bundle view to quickly find the missing messages. 

[image]

To support checkstyle, we'd just need to add a [checkstyle.xml](https://github.com/tmijail/phoebus/blob/checkstyle/checkstyle.xml) file somewhere.

---

I'm curious about what localization maintainers think. Do you prefer option 1, 2, 1+2? Or maybe some other option I haven't considered?

kasemir, 2020-08-31:
The recent change of falling back to the default language (English) is helpful, so the end user will always see _something_.
Having ways to check the localization for German, Czech, Spanish, .. is also good, but this should be an on-demand operation, performed by whoever cares about a specific language and wants to spend some time to update it.
It must not be a blocking component of the automated build, meaning our travis/... setups must not FAIL because of a missing localization.

I think a good solution right now would be to simply describe the use of the maven as well as checkstyle approach in the manual. Add the localization check to the maven files, but as an optional profile, in a way that it won't fail a build. Add a check style example. Then describe how to use them in the manual, i.e. under phoebus/docs/source/ add a "Localization" section that describes how we encourage use of the `NLS` class and `messages*.properties` files to localize text, and explain how to   use maven or checkstyle so that developers who have some time to update the localization for a language can learn how to proceed.

pavel-ch, 2020-09-02:
For me, good enough with maven report. I am only not able to do that
> ... describe the use of the maven as well as checkstyle approach in the manual. Add the localization check to the maven files, but as an optional profile, in a way that it won't fail a build ... etc

So if somebody takes care I will be grateful.
Thanks.

tmijail, 2020-09-02:
I can take a crack at it. I think that is only fair as this all originated from trying to solve one of my pain points. 
It might need some editing, though, as I don't know whether my english is good enough.

tmijail, 2020-09-21:
Closed by #1533

closed 2020-09-21 by tmijail