ControlSystemStudio/phoebus #1514: Channel tree * search is incomplete

tynanford, 2020-09-03:
Hello,

I notice that the Channel Finder search in Phoebus channel tree seems to exit prematurely when using * (get all channels)

For instance, if I search *hb and configure channel tree to show PV Status > Hostname > IocName, then I see all of our IOCs (and all PVs matching *hb).

But if I just search * and do the same configuration, I see about 20% of the IOCs. Let me know if there is other information that would be helpful. Thanks

shroffk, 2020-09-04:
Hi, 
ChannelFinder searches do have a numerical limit... by default it is 10000.
I think you can increase this number with the 'size' parameter.

The thinking was that many tens of thousands of pv's might not be best served via a UI app.

TODO: 
I will create an issue ticket in phoebus to address this.
I can add some indications that the search has hit 10k to avoid confusion.

tynanford, 2020-09-04:
Thanks for the info that makes sense.

As we started to add recsync to IOCs, I found channel table to be a really easy and quick way to see that recsync was working and the IOC made it's way to channel finder and is active. I still can do that, just have to be a little less lazy and not use * :)

closed 2020-09-04 by tynanford