ControlSystemStudio/phoebus #1515: Databrowser plot statistics

georgweiss, 2020-09-03:
I have a user requesting statistics (mean, standard deviation) for plots in the data browser. I would suggest we add another tab to show this type of data for each plot. However, I'm not sure where to put the code calculating the data: in the rtplot module or databrowser module?

kasemir, 2020-09-03:
Certainly data browser, not rtplot.
Still, this seems to be a very specific request that might make sense to that one user right now, but is not generally applicable to a tool like the data browser.
People use the data browser to look at some ADC channel in the 10 minutes, some temperature over a week, vacuum excursions with exponential behavior, motor movements.
When you look at a rotational table where the motor turns 0..360 degrees over 1 hour, of what use is the "average motor position"? For a vacuum excursion with a spike up by 3 orders of magnitude, then an exponential decay back down, of what use is the "standard deviation"?
I understand that every user would love to have the answer to their specific question de jour right now and there, but they have to understand that their questions tend to be very different, and might require exporting the data and then looking at it in Excel, Matlab, ...
If an 'Analysis' feature is added, it certainly needs to be optional and implemented in a way that it adds zero CPU load unless it's actually in use. For example, it could be an 'Analysis' tab where one can select one channel and then see mean, std. dev, ..., but by default no channel is selected and thus nothing is done at all, no added CPU load. Only when a channel is selected will its samples be analyzed. Also note that your data can contain optimized archived samples where each sample is already a min/max/average, so you'll be looking at the average of averages.

georgweiss, 2020-09-03:
My intention was not to calculate average and standard deviation in real time for each plot as it updates with new samples (whether live or from archiver). I would suggest that a new tab would present these values "on demand", e.g. on button press.

kasemir, 2020-09-03:
Yes, exactly, on demand. Technically, similar to the `SampleView`: A tab that users can optionally open, it has a list of channels, select one, ...