ControlSystemStudio/phoebus #1517: Request for Action button: on open webpage, $(pv_value) should be available

tynanford, 2020-09-04:
For the action button, there is a PV Name property.

I would like the open webpage action to be able to access the PV value from the PV name property. Is this easily possible? In this way, a base URL could be used for many buttons that open similar webpages (like links to a common wiki)

I created an example phoebus display showing what I envision. Thanks.

[open-webpage-example.zip](https://github.com/ControlSystemStudio/phoebus/files/5173004/open-webpage-example.zip)

kasemir, 2020-09-04:
An action button doesn't connect to a PV.
It does have a `$(pv_name)` macro for backwards compatibility to BOY, so that write-PV actions can then use that for a PV name. You can use macros in general, like if your want to define a base URL, you can define such a macro and then use `$(base_url)` in the open-web-page action.

But what you want to do is: Have another widget connect to a PV, and then use that PV's value for the web link. For that, your action would have to be a script somewhat like this:
```
from org.csstudio.display.builder.runtime.script import PVUtil, ScriptUtil

# Get the PV's value from e.g. text entry widget
other = ScriptUtil.findWidgetByName(widget, "NameOfThatOtherWidgetThatHasThePV")
value = other.getPropertyValue('text')

# Construct the URL using that value and combining it with whatever
url = "http://whatever/%s/whatever" % value

# Open that web page
```

The thing is that there is currently no easy way to open a webpage from a script.
I'll add that to the `ScriptUtil`

kasemir, 2020-09-04:
See pull request which adds `ScriptUtil.openWeb` and includes example script.

tynanford, 2020-09-04:
I tested out the new commit and it works great. Thanks for the quick response!

closed 2020-09-04 by tynanford