ControlSystemStudio/phoebus #1520: Question

georgweiss, 2020-09-08:
A user is requesting the possibility to configure the spinner widget to not write a value if it exceeds limits (whether configured on widget or read from PV). If configured in this manner the user should be informed somehow to clarify that the wanted value was not written. A pop-up is suggested, but this should be communicated in a less intrusive manner, e.g. red background color for a second or similar.

So our question is if and how to deal with the request. I'm asking as it would change the behavior of the widget and it would not be obvious in runtime that different spinner widgets in an OPI could be configured differently. While one widget would write the maximum value if the limit is exceeded, another would refuse to do so. Such a difference would cause some confusion.

The only other control widget where one can configure "Limits from PV" I can find is the scaled slider. However, here the user cannot specify an arbitrary value, so the use case does not apply.

georgweiss, 2020-10-08:
As discussed in joint video call October 7th, there seems to be an agreement to **not** handle writes outside the limits defined by control limit (DRVL/DRVH) in any particular fashion. For the spinner widget this means maximum/minimum value will be written by IOC and read back and presented by the widget.

closed 2020-10-08 by georgweiss

dheeraj539, 2021-02-25:
> A user is requesting the possibility to configure the spinner widget to not write a value if it exceeds limits (whether configured on widget or read from PV). If configured in this manner the user [...]

> confusion.

> 
> The only other control widget where one can configure "Limits from PV" I can find is the scaled slider. However, here the user cannot specify an arbitrary value, so the use case does not apply.

We have a similar issue where the user should be prompted with a pop-up if the input value is not within the limits. We tried to implement it using the script but we are facing a problem where the spinner gets incremented/decremented when we click on the 'OK' button on the pop-up dialog. 

If the max value is 10 and if we try to increment it above 10 then it shows a pop-up and immediately my script sets it back to 10.  But when I click the 'OK' button on the pop-up it increments the spinner value and so on..... So It's forming an infinite loop.

Any reasons why the spinner widget is behaving this way?

[image]

kasemir, 2021-02-25:
A script that prompts on values bumping limits has the following bad side effect:

Two or more people open the display.
One user, or maybe some other tool sets the value to the limit.
Now all displays will receive that value at limit and open a prompt, even on the displays where the user didn't "do" anything.
Worse: If the user has not noticed the dialog, he/she might try to use other parts of the GUI but they're now blocked because the UI is waiting for the dialog.

Another issue with adding dialogs triggered on the value is the asynchronous handling of the data.
When you move a control widget, it writes the desired value to the PV.
The PV might change to that value, or to another value, because it's really up to the code on the server side to decide what it does with the value. After some usually short time the client then receives the updated new value, which might be what you wrote or something else.

Overall, existing display tools thus simply write the value and then react to the received value, they leave any limiting, rounding, delaying, stepping to the server side.

georgweiss, 2021-02-25:
We have had extensive internal discussions at ESS on this topic. These were conducted on a more generic level, i.e. not only in the context of the spinner widget. Our conclusion and decision is to not support the requested behavior, for a number of reasons, some of which are clarified by @kasemir.

dheeraj539, 2021-02-25:
> A script that prompts on values bumping limits has the following bad side effect:
>
> Two or more people open the display.
> One user, or maybe some other tool sets the value to the limit.
> Now all [...]

Thanks!!
I agree with all the reasons why we shouldn't go with the prompts on values bumping limits.
Out of curiosity can you explain a bit more about why clicking 'OK' button on pop-up box leads to change in value of the spinner widget.

kasemir, 2021-02-25:
>...can you explain a bit more about why clicking 'OK' button ..

Not right away. You can try to debug what's happening by running 'probe' in the GUI, a 'camonitor' on the command line and adding 'print val' to your script. Then check what the PV seems to do, when the script is triggered, when the GUI updates etc.