ControlSystemStudio/phoebus #1537: Problem with CFProposalProvider

krisztianloki, 2020-09-21:
The `CFProposalProvider` class might be instantiated on the JavaFX Application Thread and when there is a problem with the network it causes hangups:

> *********************************
> *** JavaFX Application Thread ***
> *********************************
> "JavaFX Application Thread" prio=5 Id=20 WAITING on java.util.concurrent.FutureTask@3ff8fa54
> 	at java.base@11.0.8/jdk.internal.misc.Unsafe.park(Native Method)
> 	-  waiting on java.util.concurrent.FutureTask@3ff8fa54
>	at java.base@11.0.8/java.util.concurrent.locks.LockSupport.park(LockSupport.java:194)
>	at java.base@11.0.8/java.util.concurrent.FutureTask.awaitDone(FutureTask.java:447)
>	at java.base@11.0.8/java.util.concurrent.FutureTask.get(FutureTask.java:190)
>	at app//org.phoebus.channelfinder.ChannelFinderClientImpl.wrappedSubmit(ChannelFinderClientImpl.java:1243)
>	at app//org.phoebus.channelfinder.ChannelFinderClientImpl.getAllTags(ChannelFinderClientImpl.java:317)
>	at app//org.phoebus.channelfinder.autocomplete.CFProposalProvider.<init>(CFProposalProvider.java:38)
>	at java.base@11.0.8/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)

I'm not even sure if `ChannelFinderImpl.getAllTags()` is necessary; the return value is thrown away.

shroffk, 2020-10-08:
I will look into this to move the call off the UI thread...
The getTags() method was added as a check to see if it is possible to successfully connect to the service.

Are you using channelfinder?

krisztianloki, 2020-10-08:
I haven't been able to connect to our channelfinder from my development machine (always returns a 401 error) yet so I've never used it myself

shroffk, 2020-10-08:
oh, 
There are a few things you could try, 
1. ensure that the channelfinder service is accessible from the machine you are running phoebus on.
2. Make sure the preference is correctly set and  I think you should try to use the "http"
```
org.phoebus.channelfinder/serviceURL=https://localhost:9996/ChannelFinder
```

TODO: 
resolve why the channelfinder preferences are not being added to the preferences document

shroffk, 2020-10-08:
@krisztianloki This should fix it... are you able to test this before I merge it?

krisztianloki, 2020-10-09:
Yes, @shroffk , this fixes the issue. Thank you.

And it turned out that the password was wrong for the channelfinder service... I didn't really care because there is nothing interesting on the channelfinder in our lab network :)

commit bd59ecd1 (2020-10-09): Merge pull request #1562 from ControlSystemStudio/patches

fix #1537
closed 2020-10-09 by shroffk