ControlSystemStudio/phoebus #1538: Tool tip live update in Intensity widget

oksanagit, 2020-09-21:
Hi, 
Our users asked to have live update tooltip functionality on Intensity widget, which they have been relying on in their operations in CSS Studio. They would like to place a mouse on  an Intensity widget and see live updates of the intensity in this place as images coming in. After explaining about performance compromise, they said, that it would be OK to have this functionality explicitly be switched on/off. I did not discuss the max frequency, but I think l limiting to 10 Hz would be reasonable. Thank you.

kasemir, 2020-09-21:
In BOY, the image/intensity widget tool tip was used to convey information like pixel coordinates, value of that pixel, plus maybe script-enhanced detail like detector location, scattering angle etc.
That was problematic for several reasons:

 * Tool tips come and go at somewhat unpredictable times. Need to hold the mouse pointer "real still" to get the TT to appear, then it might vanish while you're still trying to read what it says
 * Users who ran CS-Studio on a Linux host from a Windows PC via some remote-X software ran into tool tip size limitations (OK, using remote-X is at best very unfortunate, but some people still insist in using it, maybe just so that they can then complain)

So for the Display Builder the tool tip is still used to show PV names, things that are cute to have but not essential. For the image widget coordinate information, refer to the Example Displays, "Image" example, and the "Cursor Info" in the lower right corner.

You basically define a cursor info PV in the image widget. That PV receives a table with coordinates, value etc. You could just display that PV in a table widget, but usually it's read by a script which then determines   detector location, scattering angle etc.  If you're not prepared to handle an "EPICS 7" type table PV, you can define separate Cursor X and Y PVs, which can be normal channel access PVs. On the SNS accelerator, those are on an IOC which then determines  detector location, scattering angle etc. and puts that into a string PV.
In any case, the cursor info is displayed in text fields next to the image, not in the somewhat erratic tool tip.

In "Crosshair" mode, the cursor PVs reflect the location of the crosshair, i.e. it's positioned where you click in the image. When "Crosshair" mode is off, the  cursor PVs reflect the current location of the cursor, i.e. it changes as you move the mouse across the image.

oksanagit, 2020-09-21:
OK, I meant cursor info, not tooltip. I use it in its current version. It does provide position, and intensity at the cursor position. But Intensity is not being updated if we leave the cursor static. Would it be possible to add this functionality? Or, it it is already there please guide me how to make it work.
Thank you. 

[image]

kasemir, 2020-09-21:
Ah, you're keeping the cursor position, but you're updating the image?
Yes, the cursor info currently only updates when you move the cursor.
Since the 'table' PV includes the 'Value', I guess it should update with image changes as well.
Will have to add that.

oksanagit, 2020-09-21:
Yes, this is correct.
Thank you.

oksanagit, 2020-09-28:
Thank you, Works fantastic!

closed 2020-09-28 by oksanagit