ControlSystemStudio/phoebus #1540: Change “Tab Height” of the “Tabs” widget doesn't work

haitao127, 2020-09-22:
When using "Tabs" widget, the default value of the "Tab Height" is 30, it can be changed to another value from "Properties", but seems doesn't work in the editor and runtime display.

kasemir, 2020-09-24:
Cannot duplicate. Seems fine on both Mac and Linux.
Original tabs example with tab height 30:

[image]

Changed to 60:

[image]

haitao127, 2020-09-27:
Sorry, I know the reason, maybe it is because of the font, because I changed the font to "[Source Han Sans](https://github.com/adobe-fonts/source-han-sans)" which can better support Chinese characters.

But if I use the default font "Liberation Sans" or other fonts, it also doesn't work well in some cases.
For example, it only works fine when the font size is smaller than 24, if I change the font size to 24 or bigger, the height will return back to the default value 30 and the setting of "Tab Height" doesn't work.