ControlSystemStudio/phoebus #1544: Failed save operations not communicated to user

georgweiss, 2020-09-24:
If a display file (and other resources?) cannot be saved because (for instance) a remote disk is unavailable, Phoebus will log the problem but leave the display editor unchanged, i.e. asterisk in tab title is still indicating dirty file. There is no alert to indicate the issue. 
In case user tries to terminate Phoebus with dirty display file, the subsequent prompt will give the user the opportunity to save. If at this point save fails, the editor will remove the asterisk giving the impression that the file has been saved successfully.

I would maybe challenge the need to to the save operations in the background, even if they do take time (and most of the times save operations should be quick). It should be easy to communicate failures, and handle the outcome in a synchronuous manner, e.g. when file save is successful, the application may be terminated. If a user wants to close the application, it will prompt before close if a file is dirty, but then not actually close once the file has been saved.

kasemir, 2020-09-24:
> I would maybe challenge the need to to the save operations in the background

Absolute necessity.
In the control room, operators run an instance of CS-Studio with many displays, ideally for months.
When they now perform file access, be it opening a new display or saving a *.plt file, that can should be quick but it's possible that for example NFS issues block it for minutes. This must not result in the complete UI being blocked.

georgweiss, 2020-09-24:
NFS issues or not, the problem remains: if file save fails user is left with dirty resource or - even worse, as described previously - left with the impression that file has been saved successfully.

shroffk, 2020-09-24:
At nsls2 we struggled with opi screens of an ill configured NFS, significant freezing... especially when we have screens with hundreds or embedded screens each with embedded screens.
These screens would have auto resizing...
Overall opening files sometimes tooks minutes and cs-studio blocking for all that time would be terrible.

So... loading files off the UI thread is **very** useful.

Now that I have given my strong opinion on this topic... @georgweiss the problem you present is a completely valid one.

> if file save fails user is left with dirty resource or - even worse, as described previously - left with the impression that file has been saved successfully.

I think we can have the file save operation expanded... so that if the user cannot save the file then we should pop up a dialog to help them save the file somewhere else where they have the appropriate authority.

kasemir, 2020-09-24:
Details of how to save are necessarily left to the application, i.e. the `DockItemWithInput` `save_handler`, because saving could mean writing a file, submitting a transaction to a database, invoking some web service etc.
That save handler is a `JobRunnable` which can report progress, support cancellation or throw an exception to indicate failure.

What's your specific issue? For the display builder editor, when a file is read-only, there's a dialog with a respective message. When I simulate a save error by throwing an exception in `DisplayEditorInstance#doSave`, this results in a "Save error" message box that details the stack trace:

[image]

Where do you get a failed save without indication?

kasemir, 2020-09-24:
> if the user cannot save the file then we should pop up a dialog to help them save the file somewhere else

That's exactly what the display editor does for the "file is read-only" case:

[image]

georgweiss, 2020-09-24:
@shroffk, I have no objections to loading displays in the background. Yes, NFS can be a pain, we clone all OPIs read-only in the control room, so we do not support editing of OPIs in the control room.

As for silent fails... My use case is this: I create a new display on share mounted over CIFS. Once the file is dirty I disconnect the file share and press save button. This creates a log entry but no dialog and the file is indicated as dirty since the save operation failed. When I try to terminate the application, I choose to save the file when prompted. Again the failure is logged, but this time the editor removes the asterisk and treats the file as saved.

kasemir, 2020-09-24:
Try to figure out why this type of failure to write the file doesn't result in an exception that would then be shown in that "Save error" dialog.

shroffk, 2020-09-24:
I think I have an example of a 'save' failing with no save error dialog.

```
2020-09-24 10:01:22 SEVERE [org.csstudio.display.builder.editor] Cannot save as C:\workspaces\phoebus-workspace\ReadOnly\read_only.bob
java.io.FileNotFoundException: C:\workspaces\phoebus-workspace\ReadOnly\read_only.bob (Access is denied)
	at java.base/java.io.FileOutputStream.open0(Native Method)
	at java.base/java.io.FileOutputStream.open(FileOutputStream.java:298)
	at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:237)
	at java.base/java.io.FileOutputStream.<init>(FileOutputStream.java:187)
	at org.csstudio.display.builder.editor.EditorGUI.saveModelAs(EditorGUI.java:537)
	at org.csstudio.display.builder.editor.app.DisplayEditorInstance.doSave(DisplayEditorInstance.java:444)
	at org.csstudio.display.builder.editor.app.SaveDisplayButton.lambda$new$0(SaveDisplayButton.java:31)
	at org.phoebus.framework.jobs.Job.execute(Job.java:50)
	at org.phoebus.framework.jobs.JobManager.execute(JobManager.java:54)
	at org.phoebus.framework.jobs.JobManager.lambda$schedule$0(JobManager.java:45)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
```

It points to this
```
    public void saveModelAs(final File file)
    {
        logger.log(Level.FINE, "Save as {0}", file);
        try
        (
            final FileOutputStream fwriter = new FileOutputStream(file);
            final ModelWriter writer = new ModelWriter(fwriter);
        )
        {
            writer.writeModel(editor.getModel());
            fwriter.flush();
            this.file = file;
            editor.getUndoableActionManager().clear();
        }
        catch (Exception ex)
        {
            logger.log(Level.SEVERE, "Cannot save as " + file, ex);
        }
    }
```

looking into it a bit more

georgweiss, 2020-09-24:
That is correct, it's new FileOutputStream throwing the exception.

kasemir, 2020-09-24:
OK, I can fix that by passing the exception up.

closed 2020-09-28 by georgweiss