ControlSystemStudio/phoebus #1556: Archive Engine

libaf, 2020-10-06:
Hi,
In the documentation of Phoebus a RDB Archive Engine is mentioned. Is this supported and is there any open repository for this?
Cheers,
Markus Kristensson
LIBAF

kasemir, 2020-10-06:
https://github.com/ControlSystemStudio/phoebus/blob/master/services/archive-engine/doc/index.rst

libaf, 2020-10-06:
In the link you provided it states that the archive engine can be built from sources or fetched as a binary from http://controlssoftware.sns.ornl.gov/css_phoebus. Could you please specify where since I couldn't find it on that link?

kasemir, 2020-10-06:
I'm very sorry that you have trouble navigating the web site. On http://controlssoftware.sns.ornl.gov/css_phoebus, look for "Nightly builds", and then you should find  archive-engine.zip in the list.

But it's also build by following the general build instructions on https://github.com/ControlSystemStudio/phoebus/blob/master/README.md, you'll find the result under services/archive-engine

libaf, 2020-10-06:
There it is! Sorry for missing that, but hanks for your help!

closed 2020-10-06 by libaf