ControlSystemStudio/phoebus #1561: Alarm datasource expansion

shroffk, 2020-10-08:
The current alarm channels are primarily suited for textupdate and textinput widgets.
We need to add more subfields to the alarm channels so that they can be more effectively used with other widgets, particularly buttons and leds.

alarm://my/alarm/node.active
Read a VBoolean
Value: true/false
Alarm: Undefined/Major/Minor/OK

Write a VBoolean
ack/unack

alarm://my/alarm/node.enable

Read a VBoolean
Value: true/false
Alarm: Undefined/Major/Minor/OK

Write a VBoolean
enable/disable

alarm://my/alarm/node.state
Return an VEnum

Read only?
alarm://my/alarm/node.desc
Return a VString
Desc and messages...if it has been defined.

shroffk, 2020-10-08:
@georgweiss @tanviash

georgweiss, 2020-10-08:
Should alarm://my/alarm/node.desc be VStringArray?

shroffk, 2020-10-08:
Processing a string array adds complication. 
I am not opposed to it but do you think it is ok if we starte with string and then based on some user feedback update it to VStringArray once we have a few use cases

georgweiss, 2020-10-08:
But if a string contains multiple parts it would in practice be processed anyway to extract the relevant portions.

kasemir, 2020-10-08:
What is the description? The alarm guidance?

shroffk, 2020-10-08:
Yes, the desc was mapping to guidance. 

@georgweiss, I was talking about how the user would have to process the array, I think for the time being I am going to skip the .desc . the other fields are far more clear and important.

kasemir, 2020-10-08:
Unclear how the dynamic behavior of alarms coming and going, and the potentially large amount of guidance maps to a display where a once-created arrangement of widgets shows alarm PVs.

An alarm config node can have one or more guidance entries. They each have a title and content. In the alarm table, you see nothing when there are no alarms, then those entries which are in alarm. Opening the context menu, you might find guidance with titles like "Overview", "Overpressure Procedure", etc.
Users are presented only with what's in alarm, and for those they can pick the guidance they need to see based on the title.
Putting all guidance titles and contents into a alarm:/path/to/node.description string array, what would you do with that?
Create a display that shows all the guidance for all the alarm nodes, and the user then reads the whole thing? Add rules to the display to show/hide guidance?

shroffk, 2020-10-08:
> Putting all guidance titles and contents into a alarm:/path/to/node.description string array, what would you do with that?

I think this is the crux of the matter...
I think I will remove the `.desc`

shroffk, 2020-11-25:
@georgweiss can we close this

georgweiss, 2020-11-25:
Yes

closed 2020-11-25 by shroffk