ControlSystemStudio/phoebus #1564: Conversion of shapes and their borders

shroffk, 2020-10-09:
BOY allowed for shaped to define borders. 
[image]

While all the 16 border options are excessive and it does not make sense to reproduce them in Display Builder, supporting a few borders which map directly to the supported `line_styles` could be very useful.

shroffk, 2020-10-09:
While exploring this issue I think I appreciate how DisplayBuilder does not have overlapping and confusing properties.
In BOY the shapes had "borders" AND "outlines"... with borders being potentially alarm sensitive.

With the conversion, I hope to stick to the property lists already established in display builder shapes. i.e. outlines.
I think it is best to preserve the broadly passive nature of the shape widgets.

kasemir, 2020-10-09:
I appreciate that you appreciate it ;-)
The borders in BOY are partially a side effect of what SWT Draw2D figures happened to offer. A 'Group' or 'Title Bar' type of border on a circle doesn't really make any sense.
Importing a 'border' of solid, dash, dash-dotted style into the line style the new rectangle makes perfect sense.

pavel-ch, 2020-10-09:
Did Boy allow setting some parameters for dash or dash-dot segment lengths? In DB, it is a fixed ratio to line (or border)  width and thus sometimes unusable (try dash-dot with line width 1):
[image]

shroffk, 2020-10-09:
@pavel-ch as far as I can see there was no such parameters in BOY.

First iteration,
[image]

TODO: 
code cleanup, move handling to support class
add same functionality to arc, ellipse, and polygon

kasemir, 2020-10-13:
I'd like to avoid too many widget-level parameters, i.e. no per-widget "Dash Length", "Dot Length", "Gap Length" parameters.
But I guess your point is not to allow arbitrary patterns or detailed customization of the dash length anyway. You're concern is simply that the dashes become invisible for certain line width.
Maybe you can find some way to scale based on the line width? Like `gap = line_width`, `dash = 3*line_width`?

pavel-ch, 2020-10-13:
Well, scaling of gaps/segments relative to line width is the current state. Changing this calculation will deny compatibility to BOY with the same calculations. Unfortunately the result of these calculations is not really good looking for "low" and "high" line width.
I would add 1 parameter which further multiplies the result of this calculation. I did not test that so far but I think this could help, and this approach keeps both BOY compatibility and a way to adjust segments somehow.
I understand that number of widget level parameters should be kept as low as possible, but I do not see any other suitable way. Maybe the parameter can be hidden in Editor until the line style is not dash/dash-dot.

kasemir, 2020-10-13:
Well, try to change the scaling. It doesn't have to be linear. Maybe `dash = 3*line_width` works for a certain range, but for small resp. large line widths the scaling needs to be different? The dot/dashes don't line up 100% with BOY anyway, since for example the starting point is different.

pavel-ch, 2020-10-13:
Ok I give a try making scale non-linear without any additional parameter and will see...

pavel-ch, 2020-10-13:
Ok looks like a simple limit:
```
        final Double seg_short = line_width > 4. || style == LineStyle.DOT ? line_width : 4.;
        final Double seg_long  = line_width > 4. ? 3. * line_width : 12.;
```
does it:
[image]

Wide lines in fact can be left as they are.
I will PR if that's ok for you.

kasemir, 2020-10-13:
Looks good to me, thanks!

shroffk, 2020-11-25:
@pavel-ch can we close this?

pavel-ch, 2020-11-25:
Yes, thanks.

closed 2020-11-25 by shroffk