ControlSystemStudio/phoebus #1605: Rectangle with action .opi->.bob translation: Action is not correcly translated.

oksanagit, 2020-10-23:
[_ip.opi.txt](https://github.com/ControlSystemStudio/phoebus/files/5426552/_ip.opi.txt)

It looks like the Action is not being translate correctly: "open filexx" instead of "Open Display". Please let me know if any more info/details is needed. Thank you.

kasemir, 2020-10-23:
??

On "Rectangle 3", the action is imported as an "Open Display" type of action:

[image]

Maybe what you see is more of a runtime problem, i.e. _invoking_ that action fails for some reason, like file cannot be resolved?

oksanagit, 2020-10-23:
I think, you are correct. I have created the action with Display builder, and when I run it, the action  is not invoked for the square with action.   It seems it should work. Thank you.
[image]

kasemir, 2020-10-23:
How do you invoke the action?

shroffk, 2020-10-23:
In BOY, the action was linked to mouse click.

[image]

With the migration to phoebus/DB...we had a discussion weather these hidden actions were good.
I think I am coming around to the idea that it might be worth educating users that they might have to do `right click`

oksanagit, 2020-10-23:
Right click is fine, users can be educated. :) However, with the way the widget is translated, it leaves the "Description" field empty, resulting in the confusing interface.
[image]
I was wondering if "Open Display" text could be added during conversion from .opi to .bob. Thank you.

kasemir, 2020-10-23:
Right, Kunal had considered implementing the "Run action on click" functionality, but we're unsure.

Fundamentally, instead of a rectangle that happens to change color with a rule, we suggest using LED widgets.
Instead of a rectangle that happens to react to click, we suggest using a button widget.
Displays with LEDs and buttons are most obvious in their expected behavior. Any new user who looks at the display for the first time has a fighting chance to guess that the button can be clicked, and the LED might change color. They stand a good chance of being auto-translated into the next display tool.
So while this type of display is not encouraged, it's understood that such displays are out there. Maybe because the designer had too much time at hand, or maybe because there's a definite need to compress as much functionality as possible into a given screen space, targeting trained operators who know which section is clickable.

As for the description, it's hard for the import to guess a description.
In the BOY case, it was simply not needed (not shown), so left empty.
Best would if you add the description in the original BOY file.
The BOY case would still work, and the import into the Display Builder would then have it.

oksanagit, 2020-10-23:
Thank you both for details and discussion.

closed 2020-10-23 by oksanagit