ControlSystemStudio/phoebus #1606: dependabot PR's

shroffk, 2020-10-23:
As everyone might have noticed we have a ton of PR's created by github helping up keep our dependencies up to date.
All the PR's are of the same nature. They are updating a single dependency junit, from 4.12 to 4.13.1

This change is trivial and I think it might be OK if we accept them...but I will be fine if we decide to reject them too.

However we have another questions to resolve.
Should we opt out of the github depandbot?
I think it might be useful to have warning and such on the github page... but I am not so sure about automatic PR's

kasemir, 2020-10-23:
Yes, I think we should update junit.
But instead of just accepting those per-pom updates, I think we should
1) Define a version property for junit, mocking, hamcrest, .. in the parent pom
2) Check if they work together, note comments in https://github.com/ControlSystemStudio/phoebus/blob/5ca634fd2325a55dc4360d6b94a69f0dc99ecbe5/dependencies/phoebus-target/pom.xml#L62
3) Then use those properties in the individual poms.

kasemir, 2020-10-23:
I'm OK with keeping dependabot enabled, but there may be times where we acknowledge, create a ticket like #1409 to follow up later, and for now ignore it.
Or need to put it some more work beyond just accepting, as I think is required for Junit

shroffk, 2020-10-23:
> Define a version property for junit, mocking, hamcrest, .. in the parent pom

 +1 I strongly prefer this way to taking a more concerted effort towards curating dependencies.

shroffk, 2020-12-03:
I have disabled the dependabot auto PR feature.
We still can use the verification and should periodically update the dependencies.

closed 2020-12-03 by shroffk