ControlSystemStudio/phoebus #1609: Crash when using ./phoebus.sh

libaf, 2020-10-27:
After a successful build I enter into phoebus-product. If run with mvn exec:java it launches fine. If run with ./phoebus.sh Phoebus crashes at startup: 

<small>
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f9a4f81d6f7, pid=2267784, tid=2267835
#
# JRE version: OpenJDK Runtime Environment (14.0.1+7) (build 14.0.1+7-Ubuntu-1ubuntu1)
# Java VM: OpenJDK 64-Bit Server VM (14.0.1+7-Ubuntu-1ubuntu1, mixed mode, sharing, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# C  [libc.so.6+0x18e6f7]
</small>

libaf, 2020-10-28:
Problem solved by changing from java-1.14.0-openjdk-amd64 to java-1.13.0-openjdk-amd64.

closed 2020-10-28 by libaf

shroffk, 2020-10-28:
Well, I think this is still an open issue if phoebus fails to run with openJDK 14.
I will try to recreate it and will possibly reopen this issue.

I do not sure I understand why the mvn execution would run and the script would not.

kasemir, 2020-10-28:
I'm launching it on Linux (RHEL 7.8), openJDK build 14+36-1471, from a 'phoebus.sh' type script:

[image]

That generally works fine.

The default phoebus.sh contains comments with suggestions, but doesn't actually set `JAVA_HOME`, `PATH`, `(DY)LD_LIBRARY_PATH`.

When you run with men exec:java and then open Help/About, does that look like its using the same  OpenJDK 14.0.1+7 that you're trying to use from the launch script? You can also check Help/About/System Properties, look for `java.*` properties, compare those to what you're trying to use from the launch script.