ControlSystemStudio/phoebus #1638: Undo/redo stack cleared on save

georgweiss, 2020-11-16:
When user saves OPI in editor, the undo/redo stack is cleared => user cannot undo after save. This is different compared to (for instance) Powerpoint where a save operation does not clear the undo stack.

I think we should maintain the undo stack even after save.

georgweiss, 2020-11-17:
I'd appreciate some feed-back in this matter as I find the current behavior to be wrong. Users might expect undo operations to still be available even after saving a file, as in RCP-based CSS.
Basically the "dirtyness" of an OPI file in edit mode is determined by the state of the undo/redo stack. This I think is wrong as it makes perfect sense to maintain the stack even after saving a resource.

pavel-ch, 2020-11-18:
From my point this is a nice-to-have feature, fine for me if this is implemented. Need to correctly implement storing the moment of Save action, to see correctly whether is the file dirty or clean when browsing the undo stack back and forth.

georgweiss, 2020-11-18:
@pavel-ch , thanks for your feed-back. But my point is that the undo stack should not determine the dirty state. Instead, if something is put on the undo stack, or removed from it, then that is a change that should set a separate dirty flag. Only a save operation should clear such a flag. A side effect would be that even if all actions are undone (i.e. undo stack is empty), the file is considered dirty. This seems to be the case in Eclipse-based CSS.

pavel-ch, 2020-11-18:
My understanding is that the file should be marked as dirty when the saved file is different from the current model state (current state = what is to be saved to a file from the current editor model state). This does not depend on emptiness of the undo stack. Traversing the undo stack, the flag has to be set or cleared according this rule.
This is what I consider the most ergonomic.
But because you implement that, the particular implementation is of your choice.

krisztianloki, 2020-11-18:
Well, I think that it would be really confusing to me if the display was shown as dirty when I know that I "undid" every change. And sometimes I just want to undo things (to check what changes I made) and I don't want to go further than what is already saved.

georgweiss, 2020-11-18:
@pavel-ch , I agree that comparing the underlying model is the solution that is conceptually the most correct approach. I could be wrong, but gut feeling this may not be that straight-forward. How should models be compared? Compute and compare hashes based on the widget hierarchy and other properties? Compare XML representations?

pavel-ch, 2020-11-18:
@georgweiss No idea how to implement that atm.

georgweiss, 2020-11-18:
I really would prefer the "compare models" approach, but not sure if I can prioritize it right now.

kasemir, 2020-11-18:
A model comparison would be overkill because it requires keeping a copy of the 'clean' model and then comparing everything on each change.

It won't be necessary because all changes in the edit happen on the UI thread, since each change is in response to mouse/keyboard interaction.
So any addition to the undo queue marks the editor 'dirty'.

At this point, saving the editor clears the undo/redo queues, and an empty queue is considered as 'clean'. That needs to be updating, separating the 'dirty' state from the queue size.

Basic idea:
Add a 'change' counter. Each undo addition increments it, each redo (=undo removal) decedents it. Saving the editor forces the change counter to zero. `dirty = change == 0`, no matter if the change counter reaches 0 because of saving or because of 'undoing' it back to zero.

georgweiss, 2020-11-18:
There is already a change counter, I added it to deal with the case where all actions on the undo stack have been consumed, but file is still dirty. I will investigate if I can use this counter for the matter at hand.

georgweiss, 2020-11-19:
Unfortunately I cannot see how a change count can fullfil all requirements:

1. Mark file as clean when undo stack has been consumed.
2. Mark file as dirty when undo stack has been consumed, but change count is >0, i.e. number of changes exceeds undo stack size.
3. Keep undo stack after save.

Model comparison will certainly add some load, but this in-memory. Also, this is edit mode, not runtime.
Using a change/dirty flag reset on save and set on each change will not work for 1, but will work for 2 and 3.

kasemir, 2020-11-19:
'dirty' means there has been a change.
So counting changes should handle it.
Could for example count up when an 'undo' element is added, down when it is removed, counting the actual attempts, even if the undo stack is full and doesn't actually grow any more but instead drops old elements.
`dirty = count != 0`.
'Save' resets `count = 0`.

georgweiss, 2020-11-19:
When a file has been modified a number of times and then saved, count is 0. When user then performs an undo action, count is decremented to -1. Next undoable change put on the stack will increment and count is 0, i.e. file is clean, which is incorrect.

kasemir, 2020-11-19:
Well, need to tweak some detail of how actions are added to the undo stack.

Save -> change count is zeroed.
Undo -> change count -1
Redo -> change count 0

Now your example:
Save -> change count is zeroed.
Undo -> change count -1
Some new change, not a 'redo': change count must be 1

Looks like logic is not simple ++, --, but closer to

undo -> --change count
redo -> ++change_count
new item added to undo stack ->    ++change_count; if (change_count == 0) change_count = 1;

georgweiss, 2020-11-20:
Yes, this will work. But it hurts a bit to maintain a "change counter" that may assume negative values...

ralphlange, 2020-11-20:
No, it won't:
Save -> change counter is zeroed.
Undo -> change counter is -1
Some new change -> change counter is 1
Undo -> change counter is 0
   _while it should be -1 as we have undone a change that stepped the counter -1->+1_

georgweiss, 2020-11-20:
Not sure I follow...
Save -> change counter is zeroed.
Undo -> change counter is -1
Some new change -> change counter is 1
Undo -> change counter is 0 _This is says that last change put on undo stack is reverted => file unchanged_

ralphlange, 2020-11-20:
Well, the file is _not_ unchanged, as you did one undo after the last save.

Conceptionally, a counter is the wrong thing.

It's a stack, and saving adds a tag to a specific stack item (after which the work was saved).
By undoing, then adding a new change, we _replace_ the top part of the stack with the new item. Which means that tagged stack item has disappeared and you can't reach that point (= change counter 0) anymore.

ralphlange, 2020-11-20:
Unless you put the undo operation itself on the stack, which - conceptionally - would directly send you into recursive hell.

georgweiss, 2020-11-20:
Still not sure what you mean. If you start with a clean file and do one single change, then undo that, the file is back to a state where it has not been modified.

ralphlange, 2020-11-20:
I start with a clean file, do an undo, do a single change, undo that change.
The counter is at 0 (clean), while my actual state is at -1 (clean plus one undo).

ralphlange, 2020-11-20:
The counter is a valid concept if and only if the undo stack does not work across save.

ralphlange, 2020-11-20:
The problem is basically a simplified version control system: only one branch, only one tag.

As soon as you allow going back before the tag, the moment you record a new commit you're done: the tagged commit is not in your branch anymore. Even if your total number of commits is the same as when you tagged the commit, your status is not the same, because you have different commits in your branch now.

georgweiss, 2020-11-20:
Ah, but the original bug report from my user is about not being able to undo after save. I would prefer to clearly separate the undo stack from the dirty state, and I'd prefer to compare contents to determine dirty state.

ralphlange, 2020-11-20:
Here's a way to make that work with the counter approach and without comparing content:

undo -> --change_count
redo -> ++change_count
new item added to undo stack -> if (change_count >= 0) ++change_count; else change_count = size(undo_stack) + 1;

The last bit makes sure that once you add a new item in the past, you'll never be clean again until you hit save.

georgweiss, 2020-11-20:
This would generate new issue tickets as the current behavior is to treat the file as clean when all elements in the undo stack have been applied. _Unless_ for the corner case where the number of changes is larger than the undo stack size and all elements in the undo stack have been consumed, a state that should define the file as dirty.

ralphlange, 2020-11-20:
But that behavior has to be changed, else you can't go into the past at all.
If you allow going into the past, only (change_count == 0) means clean.

And your corner case is exactly what I do with that last bit: make the number of changes larger than the undo stack size, so that the file is always dirty. Which is what has to be done once you add a new change to the past, as you will never be able to get back to the saved state.

georgweiss, 2020-11-20:
My point is that if I change the current behavior I will get new issue tickets. Or maybe I misunderstand you?

ralphlange, 2020-11-20:
My point is that to keep the undo stack across saving, the current behavior must change.

georgweiss, 2020-11-20:
What about discussing this in Google Meet or Zoom, @ralphlange and @kasemir ?

ralphlange, 2020-11-20:
Sure. Jitsi? Skype? GoToMyKitchen?

georgweiss, 2020-11-20:
Kitchen is fine if you bring beer.

ralphlange, 2020-11-20:
Clarified. Thanks for the chat!

closed 2020-11-25 by georgweiss