ControlSystemStudio/phoebus #1645: ChannelFinder property editor fails

shroffk, 2020-11-19:
While editing some of the demo data property values the "edit property" action is failing silently without modifying the channel.

1. verify if any error or exception is being silently dropped.
2. ensure editing actions for properties work.

shroffk, 2020-11-24:
Hao from SNS was able to determine the cause of the above problem. The missing property owner results in the request failing to be validate.

updating the channelfinder client and the property dialog to include owner information.

[image]

shroffk, 2020-11-25:
@kasemir I think this is fixed... I will reopen this issue if any problems persist.

closed 2020-11-25 by shroffk

kasemir, 2020-12-03:
Yes, better.
I'm running with ` --demo-data=1`, and `-Dspring.config.location=/path/to/my.properties` where `my.properties` has been copied from `src/main/resources/application.properties`, disabling external LDAP and instead enabling the 'embedded' LDAP.
Phoebus UI then uses 
```
org.phoebus.channelfinder/serviceURL=http://localhost:8080/ChannelFinder
org.phoebus.channelfinder/username=admin
org.phoebus.channelfinder/password=1234
```

In the table, `BR:C001-PS:1{HC}O*` locates 3 channels.
Via the context menu on one of them, can use "Add Property" to add (really change) the property "family" to value "2" (started out as 1).
Similarly, can use "Remove Property" to then remove the "family".

Wish list:

- [ ] Channel table should update. Right now need to push "Search" button to see the updated family=2 property.
- [ ] Support for removing channels
- [ ] Support for adding channels
- [ ] Use `AuthorizationService` to enable/disable actions which change the data

shroffk, 2020-12-03:
+1 on auto refresh

+1 I can easily add the `remove channel`

adding channels, I am guessing we would want to be able to add channel with tags and properties

kasemir, 2020-12-03:
Actions to add/remove, and maybe already the actions to change info somehow ask for control, i.e. authorization.

ChannelFinder handles user/password, but from the UI we pass the fixed `org.phoebus.channelfinder/username` and `password`. We know who's running the UI ("java.user"), but we don't have the associated password, so can't send that to the channel finder.
Prompt for password, or user (suggesting "java.user") and password when POSTing to channel finder?
Or use `AuthorizationService` to control access to the context menu, while still using `org.phoebus.channelfinder/username` and `password`?
```
if (AuthorizationService.hasAuthorization("cf_edit"))
{
 ... show those context menu items...
}
```