ControlSystemStudio/phoebus #1647: Backend for undo/redo stack

ralphlange, 2020-11-20:
Related to #1638: What's a good backend for the undo/redo stack?

If the stack works both ways from the clean state (and isn't cleaned on save anymore), it should probably be organized as a ring buffer, so that oldest entries are disappearing one-by-one.

To avoid usability surprises: When the change_count gets close to size(ring_buffer), there could be a warning (pop-up?) that you're soon going to overflow so can't go back to last saved state.

georgweiss, 2020-11-23:
@ralphlange , the undo stack is already organized such that oldest items are flushed when new ones are added. I have not seen any application showing a warning message when user cannot undo due to undo stack size limitation. We have made the undo stack size configurable so the default size of 50 can be overridden.

ralphlange, 2020-11-23:
Sorry - should have checked first.
I was mentioning the possibility of a warning since you reported that this was an issue for one of your users: not being able to undo all changes.

closed 2020-11-23 by ralphlange

georgweiss, 2020-11-23:
Well, users have complained about the following:
1) Undo stack size. In some cases users are fine tuning widget positions using arrow keys and each key stroke adds one item to the undo stack. So 50 items may be too low in this (odd?) case.
2) File should remain dirty when number of changes exceeds undo stack size and all items on undo stack have been consumed.
3) Undo stack shall not be flushed upon save.

ralphlange, 2020-11-23:
Good, so we'll be resolving 2. and 3. with the "undo past save point" feature.

For 1., the smart-but-complex improvement would be squashing all similar elements (e.g. moving a specific widget) into one.
(Many text editors do something similar when you're typing - instead of keeping every keystroke, they squash them into adding a word at a time.)

georgweiss, 2020-11-23:
Yes, your suggestion seems to work. 
Thanks @ralphlange for the discussion.