ControlSystemStudio/phoebus #1650: Newcomer experiences (ubuntu-18 LTS)

bfrk, 2020-11-23:
This is the first time I have tried to build a Java application. What follows is probably totally obvious to the developers (but may not be for other users), except perhaps the crash at the end.

I first cloned the git repo (at the time of writing at https://github.com/ControlSystemStudio/phoebus/commit/523fcddd12bda222408867bb049b84e7e0b2b274).

It was easy to find out which packages from my distro (ubuntu-18 LTS) I need:

`sudo apt-get install openjdk-11-jdk maven`

I then proceeded to do a `mvn clean install` but that was a bad idea. You must set JAVA_HOME, otherwise your build will fail at the very end because some plugin cannot find the javadoc binary (and maven is too dumb to cache the intermediate results and thus will do everything again). On my machine the correct setting is

`export JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64`

Then you can run

`maven install`

and it actually succeeds. So far so good. Doing `cd phoebus-product` and then

`mvn exec:java`

unfortunately crashes:

```
2020-11-23 10:35:40 INFORMATION [org.phoebus.product.Launcher] Phoebus (PID 17302)
2020-11-23 10:35:43 SCHWERWIEGEND [org.phoebus.framework.nls.NLS] org.csstudio.trends.databrowser3.Messages contains superflous message 'NewPlotTargetFileNotDirectory'
2020-11-23 10:35:43 INFORMATION [org.phoebus.channelfinder.ChannelFinderClient] Creating a channelfinder client to : http://localhost:8080/ChannelFinder
2020-11-23 10:35:44 SCHWERWIEGEND [org.phoebus.framework.preferences] No default setting for preference org.phoebus.logbook.ui/is_supported
2020-11-23 10:35:44 INFORMATION [org.phoebus.applications.display.navigation.DisplayNavigationViewApp] Loading Display Navigation View
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f432c6685b9, pid=17302, tid=17392
#
# JRE version: OpenJDK Runtime Environment (11.0.9.1+1) (build 11.0.9.1+1-Ubuntu-0ubuntu1.18.04)
# Java VM: OpenJDK 64-Bit Server VM (11.0.9.1+1-Ubuntu-0ubuntu1.18.04, mixed mode, sharing, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# V  [libjvm.so+0x8fb5b9]
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/ben/scratch/phoebus/phoebus-product/hs_err_pid17302.log
#
# If you would like to submit a bug report, please visit:
#   https://bugs.launchpad.net/ubuntu/+source/openjdk-lts
#
zsh: abort      mvn exec:java
```

BTW, I did see a window opening before the crash.

georgweiss, 2020-11-23:
The error report file `/home/ben/scratch/phoebus/phoebus-product/hs_err_pid17302.log` may contain valuable information.

bfrk, 2020-11-23:
I tried to attach the log but github is stubborn and doesn't let me, even if I pack it with zip.

bfrk, 2020-11-23:
Ah, I needed to allow another script host. Here is the log:

[hs_err_pid17302.zip](https://github.com/ControlSystemStudio/phoebus/files/5582216/hs_err_pid17302.zip)

georgweiss, 2020-11-23:
From the error file and your description: looks like a graphics issue to me.
For what it's worth, can you reproduce using OpenJDK 14?

bfrk, 2020-11-23:
I haven't had any graphics issues before with this machine (a thinkpad T430). My system is up-to-date and uses the standard Intel graphics driver. Trying jdk-14 will be difficult for me since jdk-11 is the most recent one offered by my distro.
A colleague pointed me to a binary distribution (4.6.5) and that doesn't crash. So this appears to be a regression...

georgweiss, 2020-11-23:
The reference to "graphics" is about Java stuff (more specifically JavaFX). In any case, the Java version (11 vs 14) should actually not matter. If the binary version solves you issue, then fine. But if you wish to be able to build and run latest code, can you verify:
1) `$JAVA_HOME/bin/java -version` echoes the expected Java version.
2) Try the phoebus.sh script bundled in the package generated by the build. It's also available in the phoebus-product module. Make sure paths are correct for your environment.
3) Alternatively, from `phoebus-product/target` launch like this: `$JAVA_HOME/bin/java -jar product-<version>.jar`

Of course, you can omit `$JAVA_HOME/bin` provided you have java in your PATH and that it is Java 11 (or later).

bfrk, 2020-11-23:
`$JAVA_HOME/bin/java -version` gives me `openjdk version "11.0.9.1" 2020-11-04` which looks good. And your method (3) to start the program seems to work, thanks! (I am still wondering why `mvn exec:java` crahes like that.)

bfrk, 2020-11-23:
Unfortunately the menu bar is unusable. Normal buttons in windows work, but when I click an item, a menu pops up and it disappears as soon as I release the butten. That's okay, many toolkits work like that. But I cannot select anything. The submenus don't open, and when I release the mouse button over, say, File->Open, nothing happens.

bfrk, 2020-11-23:
That behavior is the same in the released version.

georgweiss, 2020-11-24:
This is news to me. What window manager are you using? Default on Ubuntu should be Gnome 3.

bfrk, 2020-11-24:
I am using xfce with xmonad as the window manager. After consulting with my colleagues it appears that this problem only happens with tiling WMs. I guess this is a problem originating with JavaFX and not the fault of CSS/phoebus.

bfrk, 2020-11-24:
It seems others have had similar problems with JavaFX and i3 WM: https://bugs.openjdk.java.net/browse/JDK-8251240
There seem to be no tricks floating around to make it work. Thankfully I found a way to access the menu: I can hit Alt key, which selects the File entry in the menu bar, and then navigate the menu with arrow keys.

georgweiss, 2020-11-24:
Do you have the possibility to check the behavior under Gnome on Ubuntu?

bfrk, 2020-11-25:
No, not with Gnome. Two of my colleagues also use ubuntu 18, but with xfce and its default WM. No problems there. Another one works with MacOS and tells me the problem exists there, too, at least initially after starting CSS, but then disappears once another application's window is brought on top.

georgweiss, 2020-11-25:
The issue on Mac is discussed in #1092. But based on your  problem description I'd say the issue on Mac is a different one.

bfrk, 2020-11-25:
Perhaps. I asked another colleague who uses i3 on Fedora but he hasn't reported back yet.

georgweiss, 2020-11-25:
Can we close this?

bfrk, 2020-11-25:
Yes. It is unfortunate that JavaFX has these problems but this not CSS's fault.

bfrk, 2020-11-25:
Ah, well, I forgot about (a) the crash with `mvn exec:java` with the git clone and (b) it would be nice if the installation instructions could mention that JAVA_HOME must be set.

georgweiss, 2020-11-25:
Seems `mv exec:java` is problematic also on (at least) Mac OS. Will investigate and update instructions.

georgweiss, 2020-11-26:
I've clarified instructions in README, check PR #1658 
Your observation on `mvn exec:java` is interesting, turns out it's an issue on all platforms, at least with Java 11+. Since launching Phoebus is quite simple using plain `java -jar`, I think it suffices as a hint. Plus there are launch scripts available.

closed 2020-11-27 by georgweiss