ControlSystemStudio/phoebus #1654: organization of app

shroffk, 2020-11-25:
I need to add 2 additional modules associated with the emails (latter logbook/olog too)

Currently we have a set of applications directly under app and a rather consistent naming scheme too

dir: /app/email
name: app-email

But it is not possible for adding new modules here.
Would it be ok to have the following.

dir: /app/email (parent project/module)
name: app-email

dir: /app/email/util (the main application code... I am open to a better name than util, but want to avoid app-email-email)
name: app-email-util

dir: /app/email/simple-context-menu (optional module for allowing site specific email configuration)
name: app-email-simple-context-menu

dir: /app/email/rich-context-menu  (optional module for allowing site specific email configuration)
name: app-email-rich-context-menu

This is in keeping with the established rules of both dir structure and naming even though some of the names do seem a bit long. With more changes I have planned to reduce the need to inclusion and exclusion preferences clarification and approval of the above format would be good.

georgweiss, 2020-11-25:
As for the structure I have no objections.
Naming is a different matter... app-email-simple-context-menu is a bit misleading: is the module context menu stuff only? How about:
* app-email-common
* app-email-plain
* app-email-rich-text

shroffk, 2020-11-25:
Well, for the time being the plain and rich have adapters for databrowser ( I am adding for opi runtime too)
I can remove the context menu part since it has the adapters only

I like common more than util...

@kasemir any thoughts

shroffk, 2020-11-25:
@georgweiss any idea about organizing the olog peices - the old olog client and the new olog client

/app/logbook/olog/client
/app/logbook/olog/es-client
/app/logbook/olog/ui ( viewer and editor )

georgweiss, 2020-11-25:
Isn't this the structure we have currently, save for naming?

georgweiss, 2020-11-25:
Sorry, you are looking from "olog root" perspective, right?
Then why not, separate clients as that could be isolated. But what about ui stuff upon which elog depends?

shroffk, 2020-11-25:
> Sorry, you are looking from "olog root" perspective, right?
> Then why not, separate clients as that could be isolated. But what about ui stuff upon which elog depends?

the general ui stuff remains where it is, the /olog/ui module will have the new olog specific pieces

kasemir, 2020-12-01:
As for email, note that there are also headless tools that send email.

core-email is supposed to hold just the basic email support.
app-email is supposed to hold the UI code.

In reality, core-email only has EmailPreferences.java.
app-email has an empty EmailService.java.
The actual code to create a message with title & text, and to send it out is for example inside the EmailDialogController.java.

One good update would be to move EmailService.java into core-email, and in there implement the common createMessage(title, text, ..) code which can then be used by both headless code (e.g. alarm server) and the UI in app-email. On top of that, app-email can split further into plain, rich-text etc., but first I'd update the basic send-email functionality,

shroffk, 2020-12-01:
Ok, I think this should become a separate Issue.

+1 on all 3 points

Actually, if we fix the email modules like you describe the it automatically resolves my question too

shroffk, 2020-12-03:
The reorganization of the email and olog modules is complete.
Might need to reopen as new modules are added

closed 2020-12-03 by shroffk