ControlSystemStudio/phoebus #1656: Databrowser views and standard deviation

willrogers, 2020-11-25:
We find in CS-Studio that the area and line plots that include min, max and standard deviation are confusing. It's hard to see how adding standard deviation to the plot helps anyone.

I think it is the same in Phoebus.

We would propose just not to plot standard deviation. Would anyone object to this? We could make it a preference but there are already lots of preferences.

cc @rjwills28.

georgweiss, 2020-11-25:
Phoebus Databrowser does nor plot the standard deviation, it's shown in the statistics tab only.

willrogers, 2020-11-26:
I mean when the request is optimised because there are too many points to retrieve in a reasonable amount of time. Here's an example:

[image]

That section in the middle is supposed to show min, max, mean and mean += standard deviation for a 'bin', but I find it confusing. We are separately working on a better algorithm for the Archiver Appliance as well.

rjwills28, 2020-11-26:
The plotting of the standard deviation occurs here in the code: https://github.com/ControlSystemStudio/phoebus/blob/master/app/rtplot/src/main/java/org/csstudio/javafx/rtplot/internal/TracePainter.java#L135

georgweiss, 2020-11-26:
Sorry, I got it totally wrong.

willrogers, 2020-11-27:
@kasemir what do you think about this?

shroffk, 2020-11-30:
Hi will, 
I feel like plotting(area and lines) the standard deviation usually causes confusion... mean, min, and max are usually enough from my experience. 
Especially when we have more than one signal being plotted.

However, showing the data in the sample viewer is definitely useful.

So for visualization maybe we should have add another type? area, stat_area(area + std), lines, stat_lines(lines +std)

willrogers, 2020-11-30:
We could add more configuration, but I'm not sure plotting the stddev is ever useful.

The data would still be there in all the other views, just not on the plot.

kasemir, 2020-11-30:
Historically, the plotting of stddev was simply added because the "VStatistics" can carry that information, so I felt compelled to represent it _somehow_.
Unclear to which extend the various archive data sources actually set the stddev.
The RDB data source that I'm using cannot fill it, so it leaves the std dev as zero.

I see at least 3 options:

The archive appliance data source can set the VStat.stddev=0. Then it won't cause any lines to be drawn. But I think you like to look at that in the 'samples view', so maybe best to still report the stddev for archive data sources that can provide the info.

Have more options in the plot type as Kunal suggested. So far, having to explain the min/max area outline already takes some time because nobody ever reads the description in the online help. For that reason I'm not too eager to add more options, which even fewer people will ever understand and never appreciate.

So that leaves Will's initial suggestion to simply remove the std dev from the plot code. Could be a preference option, defaulting to `show_std_dev=false`, in case there's anybody who wants to keep it around as an option, but I'd be OK with simply removing it for simplicity.

shroffk, 2020-11-30:
> So that leaves Will's initial suggestion to simply remove the std dev from the plot code. Could be a preference option, defaulting to show_std_dev=false, in case there's anybody who wants to keep it around as an option, but I'd be OK with simply removing it for simplicity.

+1 for wills initial suggestion. My suggestion was only if anyone really wants the std plotted (I don't use it) 
As long as it is kept in the sample view my preferred solution is to remove it from the plot.

willrogers, 2020-11-30:
OK, let's remove it. It's easy to put it back with a preference option if anyone notices and would like it.