ControlSystemStudio/phoebus #1660: Display Builder WidgetRuntime - Widget startup not completed

pavel-ch, 2020-11-29:
Another threading issue occurs on my system.
Configuration: on a display, three instances of an Action Button-like site specific widgets are configured, all three with two actions, one of them is a call to an external Jython script file (the same for all three). According to my investigation, the issue has nothing to do with the site specific code in the widget, but with the common part.
What happens:
**Rarely** we experience this exception when closing the display or executing widget action:
```
2020-11-29 00:26:08.248 WARNING [org.csstudio.display.builder.runtime] Runtime startup not completed for Widget 'HMI Alarmové tlačítko_9' (hmi_alarm)
java.lang.Exception: Stack trace
	at org.csstudio.display.builder.runtime.WidgetRuntime.awaitStartup(WidgetRuntime.java:360)
	at org.csstudio.display.builder.runtime.WidgetRuntime.stop(WidgetRuntime.java:441)
	at org.phoebus.hmi.display.runtime.HmiWidgetRuntime.stop(HmiWidgetRuntime.java:224)
	at org.csstudio.display.builder.runtime.RuntimeUtil.stopRuntime(RuntimeUtil.java:167)
	at org.csstudio.display.builder.runtime.RuntimeUtil.stopChildRuntimes(RuntimeUtil.java:193)
	at org.csstudio.display.builder.runtime.internal.DisplayRuntime.stop(DisplayRuntime.java:64)
	at org.csstudio.display.builder.runtime.RuntimeUtil.stopRuntime(RuntimeUtil.java:167)
	at org.csstudio.display.builder.runtime.internal.EmbeddedDisplayRuntime.lambda$0(EmbeddedDisplayRuntime.java:30)
	at org.csstudio.display.builder.runtime.internal.EmbeddedDisplayRuntime.stop(EmbeddedDisplayRuntime.java:50)
	at org.csstudio.display.builder.runtime.RuntimeUtil.stopRuntime(RuntimeUtil.java:167)
	at org.csstudio.display.builder.runtime.RuntimeUtil.stopChildRuntimes(RuntimeUtil.java:193)
	at org.csstudio.display.builder.runtime.internal.DisplayRuntime.stop(DisplayRuntime.java:64)
	at org.csstudio.display.builder.runtime.RuntimeUtil.stopRuntime(RuntimeUtil.java:167)
	at org.csstudio.display.builder.runtime.ActionUtil.handleClose(ActionUtil.java:196)
	at org.csstudio.display.builder.runtime.app.DisplayRuntimeInstance.disposeModel(DisplayRuntimeInstance.java:446)
	at org.csstudio.display.builder.runtime.app.DisplayRuntimeInstance.onClosed(DisplayRuntimeInstance.java:499)
	at org.phoebus.ui.docking.DockItem.handleClosed(DockItem.java:618)
	at org.phoebus.ui.docking.DockItemWithInput.handleClosed(DockItemWithInput.java:436)
	at org.phoebus.ui.docking.DockItem.close(DockItem.java:653)
	at org.phoebus.ui.docking.DockStage.closeItems(DockStage.java:245)
	at org.phoebus.ui.application.PhoebusApplication.lambda$46(PhoebusApplication.java:1117)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$10(PlatformImpl.java:428)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$11(PlatformImpl.java:427)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:96)
	at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
	at com.sun.glass.ui.win.WinApplication.lambda$runLoop$3(WinApplication.java:174)
	at java.base/java.lang.Thread.run(Thread.java:834)
```
The exception is caused by `WidgetRuntime.awaitStartup()` method, which waits for `CountDownLatch started`. This latch is unblocked at the end of `startScripts()` function, which is populated in a background job from Runtime's `start()` function.

What I see is that `startScripts()` is invoked (via helper log messages), but never returns in these cases. That means the background job is stuck somewhere inside the `startScripts()` and never unblocks the latch. Because of the widget configuration, no scripts nor rules are configured, but a script action is there. I suppose there is a kind of deadlock between background jobs running `startScripts()` on those three widgets, maybe inside the action script compilation chain of function calls. Unfortunately this happens rarely, so I need a lot of restarts to see the issue (the issue is showing mostly on startup, when opening the display runtime from memento).
I will continue with investigation.
Could you also think about that @kasemir ? Thanks.

pavel-ch, 2020-11-30:
After a number of experiments (I am not able to see the issue when debugger is active for a direct investigation) my suspicion concentrated to the `org.csstudio.display.builder.runtime.script.internal.JythonScriptSupport.addToPythonPath(String)`. I tried to move the first line - getting the Python `SystemState` - into the `synchronized` guard, due to my suspicion of a threading problem in the Jython internal initialization phase.
The issue seems to disappear, or minimally I am not able to see it in a series of tests.
So I would propose, as this change should not harm anything, to keep my change and see what happens in a long run. I admit that the change is a little bit empiric, but it seems to solve the issue for me.
PR follows.

pavel-ch, 2020-11-30:
Okay, this maybe changed some probability to see the issue, but the issue itself is still there. I need to simulate a heavy system load on my test computer (Windows 10) and the probability increases. Digging continues.

pavel-ch, 2020-11-30:
I identified the particular point of stuck: The first failed widget script compile job gets stuck on:
`org.csstudio.display.builder.runtime.script.internal.JythonScriptSupport.addToPythonPath(String)` at line:
`final int index = paths.indexOf(path);`
All following failed jobs get stuck logically on `synchronized (JythonScriptSupport.class)`.

At the end, I was happily able to catch the deadlock in the debugger. See attached printscreen.
Not clear to me by what mechanism is waiting DisplayRuntime37 thread. Others seem to wait for it: DisplayRuntime36 and DisplayRuntime wait for JythonScriptSupport.class, and ScriptSupport for PyList.
@kasemir do you see something? 
Thanks.
[image]

kasemir, 2020-11-30:
Looking at the Jython-standalone-2.7.2.sources.jar, most PyList methods are synchronized.
That means if we also lock on JythonScriptSupport.class, we need to avoid a deadlock by first locking the JythonScriptSupport.class, then touching the 'path' PyList.
With your proposal to move the `paths = python.getSystemState().path` inside the sync section, that should be the case.

kasemir, 2020-11-30:
> Not clear to me by what mechanism is waiting DisplayRuntime37 thread. 

Exactly, that's the question.
It's holding the JythonScriptSupport.class and the PyList.path locks, so those other threads which want to execute scripts that have already been compiled need to wait. Fine.
But what's keeping DisplayRuntime37, which is in the process of compiling yet another thread, from progressing?

It's stuck in PySystemState.getCodecState():
```
 public synchronized codecs.CodecState getCodecState() {
        if (codecState == null) {
            codecState = new codecs.CodecState();
            try {
                imp.load("encodings");
```

We've seen another deadlock related to getCodecState(), #1310

kasemir, 2020-11-30:
I'm not seeing the complete cycle of threads, locks they hold, locks they'd like to take, and how that deadlocks.

But one piece could be this:

"ScriptSupport" thread us in PythonInterpreter.exec, executing some script, currently inside some "imp.importFrom".
It thus potentially owns some 'import' lock that doesn't show up, and it wants PyList (id=490).

Meanwhile, "DisplayRuntime37" wants to compile some new script. It owns that PyList (id=490), and "imp.load" it trying to get some ReentrantLock, which I assume is some global import lock.

The fix might be to place a `synchronized (JythonScriptSupport.class)` in the execution of scripts:
```
   public Future<Object> submit(final JythonScript script, final Widget widget, final RuntimePV... pvs)
    {
         ...
        return support.submit(() ->
        {
            // Script may be queued again
            removeScheduleMarker(script);
            try
            {
                synchronized (JythonScriptSupport.class)
                {
                    python.set("widget", widget);
                    python.set("pvs", pvs);
                  logger.log(Level.INFO, () -> "Exec " + script + " for " + widget + " in " + python + ", locals: " + python.getLocals());
                  
                  // **** EXECUTION NOW INSIDE THE LOCK?! ********
                  python.exec(script.getCode());
                }
```

So we lock
1) When compiling scripts
2) When executing scripts

Turns out 1 was needed because otherwise the 'path' gets messed up.
2 wouldn't really make much of a difference because the scripts for one display are already executed sequentially by one thread. So no change in script execution, but we will now either compile a script (from runtime startup thread) or execute (from script support thread), not both compile and run at the same time.

kasemir, 2020-11-30:
Can you try the script_1660 branch?
If you still get the issue, could add `synchronized (JythonScriptSupport.class)` to the complete `JythonScriptSupport.compile()` instead of just the addToPythonPath called from within there.

kasemir, 2020-11-30:
> .. long running scripts ..

Right.. Scripts within one display are already serialized, since it's one executor per display.
But that still allows scripts for the various displays to run in parallel.
.. yet their 'path' is all one, and the 'import' lock is all the same ..

May need a different lock to get the following:

 Can only compile one script at a time,
 can not run any script while compiling scripts,
 but may run several scripts in parallel (one per display).

kasemir, 2020-11-30:
I assume your deadlock is related to having many widgets with scripts, and the PVs send an update very soon.
So while you're still compiling scripts, some already start to run..
Any way to duplicate that with local PVs?

pavel-ch, 2020-11-30:
Yes exactly, I see executing scripts while still compiling inside a display.
The display I experience this issue has not got too many scripts or rules, but maybe system load can add a probability to see this.
I give a try to your branch and will see.
Can try with local PVs, but later on. For that we need some fast simulated PVs with own thread. Plus set of display widgets with rules or scripts, plus some widgets with either script actions or scripts using script files, to catch simultaneous exec and compile including path modification.

kasemir, 2020-11-30:
That branch will only allow one script at a time, globally, so it's not the solution.
Need some more time to think about it.

pavel-ch, 2020-11-30:
Why not to simply synchronize like in your branch, but on JythonScriptSupport instance instead of the class? There is one instance per display. Is there any other global limitation?

kasemir, 2020-11-30:
I don't have a solution, but I think the issue is down to locks purely inside Jython, and again based on getCodecState(), #1310

One thread compiles a script, calls addToPythonPath, path.indexOf synchronizes, and then internally calls PySystemState.getCodecState, which ends up calling "imp" which wants the 'import; lock.

Meanwhile, another thread runs a script. That script happens to contain "import xxx", which takes the import lock, and then tries to traverse the path, i.e. synchronizes on it.

The order for DisplayRuntime37 is 1) Lock 'path' 2) Try to take the import lock.
The order in ScriptSupport is 1) Take the import lock 2) Try to lock 'path'.

Those orders are inverted, and both are happening inside the Jython code without us holding any additional lock that would matter here.
What could help is to assert that getCodecState() has been called early on, since it will only call `imp.load("encodings")` once. If we place a dummy call to getCodecState() in the section that creates the interpreter, it will be initialized and not run into this exact deadlock ... I think.

pavel-ch, 2020-11-30:
The solution suggested in my tweet above seems to run fine for me, using synchronize(this) instead of synchronize(JythonScriptSupport.class). Covers the issue by not allowing the concurrent run of exec and path.indexOf, while maintaining independent scripts execution on different displays. But I have no clue about possible side effects.

kasemir, 2020-11-30:
While the PythonInterpreter is one per display, they share the same global path. That's why all code which touches the path locked on a global JythonScriptSupport.class, not this.
But if the intent is really to guard the path, one could `synchronized (path)`.
Then again I think the actual deadlock does not involve any of our locks, it's purely within the jython lib.
Calling `python.getSystemState().getCodecState()` at interpreter creation would avoid that being called later.

pavel-ch, 2020-11-30:
Well. In the meantime I tried your suggestion with the very original code and `state.getCodecState();` called at the end of `JythonScriptSupport.init()`. Seems to work, too. I will do a more complete testing tomorrow.

kasemir, 2020-11-30:
Try the script_1660 branch.
It forces python.getSystemState().getCodecState() when the interpreter is created, which should avoid the `imp.load("encodings")` that caused the deadlock from the snapshot.

Otherwise it actually reduces locking: Locks on `path` when that's modified for compilation. That's actually the same `path` instance for all interpreters, so it's still a global lock, but more granular than `JythonScriptSupport.class`.

When executing a script, it doesn't lock at all.

If we find that locking on `this` is indeed needed for execute & compile, we can add that, but maybe it's not necessary.

pavel-ch, 2020-11-30:
Ok I will test on your last code, tomorrow morning CET :-) Thanks for help!

pavel-ch, 2020-12-01:
The modifications in the script_1660 branch are working well for me.
I closed my last PR and ask you @kasemir to merge these changes to master. Thank you very much for cooperation!

pavel-ch, 2020-12-01:
Perhaps being into Jyhon related code, somebody with Eclipse knowledge could help me:
The Jython related code for example in JythonScriptSupport.java appears as error in Eclipse code editor. But in the Project Explorer looks fine, compiles fine in Eclipse and the jython-standalone-2.7.2.jar is on the classpath through phoebus-target project. I am using phoebus repository as-is. Does anybody know how to set it correctly up?
[image]

kasemir, 2020-12-01:
That would be one question about Eclipse that I don't understand, either.
Right now, the jython imports all resolve fine for me:

[image]

On the other hand, I always see similar warnings for anything related to XML/DOM/SAX packages

pavel-ch, 2020-12-02:
Closed by #1663. Thanks.

closed 2020-12-02 by pavel-ch