ControlSystemStudio/phoebus #1667: Auto update download issues

tynanford, 2020-12-01:
Our users have seen issues with 2 related things with the auto updater.

1. While the update is in progress, the update button still appears, which some users have pressed again and then it launches a second update
2. When they try to cancel the update, the download continues (not sure how important this one is if the first issue isn't there)

kasemir, 2020-12-01:
Hiding the button is easy and makes a lot of sense.

The cancellation is iffy. Technically, I'm using some library code that you can't gracefully cancel. Even if you succeeded, would then have to remove the pieces. It's acknowledged that one can currently not cancel an ongoing update. If the update doesn't complete, be it for some error or because the 'cancel' actually stops the 'unzip' phase after the download, you're left with a broken setup and have to re-install.

Local feedback seems to be that the advantages outweigh the issues. Generally, at least with Windows and Mac OS, the 'Update' works and then saves a lot of time over a manual re-install.

tynanford, 2020-12-01:
This all sounds good, thank you. Our local feedback has also been very positive with this update feature (and it is very nice for maintainers)

closed 2020-12-02 by tynanford