ControlSystemStudio/phoebus #1669: build java doc error

forestmojo, 2020-12-02:
Followed the steps to build javadoc as:
# Build the Javadoc, i.e. html files to be included in the manual
( cd phoebus/app/display/editor; ant -f javadoc.xml clean all )
but there are many errors such as :
error: package org.csstudio.display.builder.model.properties does not exist
error: static import only from classes and interfaces

shroffk, 2020-12-03:
Can you try the following in the top folder.

```
$ mvn clean verify -DskipTests=true -Psphinx -N
```

kasemir, 2020-12-03:
Update docs/README.md to show `mvn clean verify ..-Psphinx` instead of example without `clean`?

shroffk, 2020-12-03:
copy pasted the clean but like you point out... it is not needed

forestmojo, 2020-12-03:
hi, shroffk:

Another question about olog, on https://olog-es.readthedocs.io/en/latest/, it says “ Detailed Installation Instructions: Install Olog-es.” , but the link is broken.  Would you please correct this?
Or does phoebus support elog now?

> 
> 
> Can you try the following in the top folder.
> 
> ```
> $ mvn clean verify -DskipTests=true -Psphinx -N
> ```

shroffk, 2020-12-15:
@forestmojo thanks for this catch
Fixed the link 
https://github.com/Olog/olog-es/commit/4dec6030c5d3ea0f6a48f5aa03bf93e26f548288

Can we close this issue

closed 2020-12-16 by forestmojo