ControlSystemStudio/phoebus #1671: Display Buider WidgetRuntime - reducing number of background jobs on display start

pavel-ch, 2020-12-02:
When debugging WidgetRuntime startup hunting deadlocks, I have noticed a huge number of background jobs (threads) runned simultaneously at display startup, executing `org.csstudio.display.builder.runtime.WidgetRuntime.startScripts()` code.
I observed with my display up to 200 simultaneous threads peak. Some of them even reused more times.
This could be one of reasons of a relative slow startup of some displays.
What I can see, too, is that most of this threads do effectively nothing: if displays are created according to recommendations and rules and scripts are rarely used, the `startScripts()` does only one thing: unblocks the `started` latch at the end.
I would propose to check existence of scripting BEFORE running the thread with  `startScripts()`. This would radically decrease the number of threads on display startup. Even that threads are from a pool and should be a bit cheaper, there must be a lot of work to synchronize etc.

The code (begins at the end of `org.csstudio.display.builder.runtime.WidgetRuntime.start()` line237):
```
        ...
        widget.propClass().addPropertyListener(update_widget_class);

        if (hasScripts())
            // Start scripts in pool because Jython setup is expensive
            RuntimeUtil.getExecutor().execute(this::startScripts);
        else
            started.countDown();
    }
    
    /** Are Scripts or Rules defined for Widget? */
    private boolean hasScripts()
    {
        if (widget.propScripts().getValue().size() > 0)
            return true;
        if (widget.propRules().getValue().size() > 0)
            return true;
        final List<ActionInfo> actions = widget.propActions().getValue().getActions();
        if (actions.size() > 0)
        {
            for (ActionInfo action_info : actions)
            {
                if (action_info instanceof ExecuteScriptActionInfo)
                    return true;
            }
        }
        return false;
    }

```

I did not do any profiling so far, but this must be better.

georgweiss, 2020-12-02:
On the topic of display load times: one of my users suggested the idea of persisting pre-compiled scripts as compilation is slow compared to execution (at least for Javascript). In my view it could be worthwhile exploring, but the cost may turn out to be too high, e.g. persisting and loading binary data, version incompatibilities in script engine library, hard dependency of script engine library etc.

kasemir, 2020-12-02:
Go ahead on the `if (hasScripts(...` idea, which looks like an easy win.

Regarding the recompilation, check this `JythonScriptSupport` comment:
```
            // Disable cachedir to avoid creation of cachedir folder.
            // See http://www.jython.org/jythonbook/en/1.0/ModulesPackages.html#java-package-scanning
            // and http://wiki.python.org/jython/PackageScanning
            props.setProperty(RegistryKey.PYTHON_CACHEDIR_SKIP, "true");

            // By default, Jython compiler creates bytecode files xxx$py.class
            // adjacent to the *.py source file.
            // They are owned by the current user, which typically results in
            // problems for other users, who can either not read them, or not
            // write updates after *.py changes.
            // There is no way to have them be created in a different, per-user directory.
            // C Python honors an environment variable PYTHONDONTWRITEBYTECODE=true to
            // disable its bytecode files, but Jython only checks that in its command line launcher.
            // Use the same environment variable in case it's defined,
            // and default to disabled bytecode, i.e. the safe alternative.
            if (System.getenv("PYTHONDONTWRITEBYTECODE") == null)
                Options.dont_write_bytecode = true;
            else
                Options.dont_write_bytecode = Boolean.parseBoolean(System.getenv("PYTHONDONTWRITEBYTECODE"));
```

You could move the `..setProperty(...PYTHON_CACHEDIR_SKIP, "true")` down to go with `Options.dont_write_bytecode` and thus offer a way to re-enable caching of the precompiled scripts, but be prepared for midnight calls because some script sudden fails to run as a result of some other user having created the pre-compiled version, a change to the script suddenly having no effect etc. The default should this remain as is.

georgweiss, 2020-12-02:
The pre-compilation idea was suggested to deal with the (expensive) task of compiling scripts on OPI launch. So, when the OPI file is saved, scripts would be pre-compiled and saved as byte code in the xml file. Then, at launch, the byte code would be loaded from xml and ready for execution. 
I'm not saying this is a good idea, there are probably a number of pitfalls, but fact remains that an OPI may explicitly or implicitly (rules) refer to a large number of scripts that need to be compiled when the OPI is loading. Whether loading byte code from xml is more efficient is an open question.
Caching is maybe a different angle...

kasemir, 2020-12-02:
Again jython already supports caching, re-enable it if you want.

Still unclear why this is such an issue in the first place.
The use of scripts and rules should be limited to begin with, unless you plan to redo/port/update your displays every 5 years.
If you do use scripts/rules for special cases, the bulk of the display tends to load and show quickly. Compared to BOY, nothing freezes. Then a few seconds later the scripts start to function. Yes, there is a noticeable delay until the scripts start, but is that an actual practical issue?

georgweiss, 2020-12-02:
The underlying issue is users' expectations: freedom to be creative with scripting, use scripts to work around poorly implemented IOCs, and to not see UI delays even on single core VM on Windows.
I'm not in favor of pre-compilation, but I think it has some value to be able to say that we have discussed it.

pavel-ch, 2020-12-02:
If the scripting cache discussion is about to continue, I would suggest creating own issue.
Closing this one via #1675

closed 2020-12-02 by pavel-ch