ControlSystemStudio/phoebus #1674: domain for phoebus

shroffk, 2020-12-02:
The cs-studio/epics community should own the org.phoebus domain.

 - We could publish a website which would serve as a landing page for new users and developers.
 - This would allow us to publish phoebus modules (core, app, services) on maven repositories like sonatype.

In the past this domain was available at an exorbitant price in the order of 10k EURs, however recently the price has decreased significantly and it is available 1500EUR. 

https://sedo.com/search/details/?partnerid=14456&language=us&domain=phoebus.org&origin=parking&utm_medium=Parking&utm_campaign=template&utm_source=3000

tmijail, 2021-07-25:
Hello. I just watched your recent talk (which was great by the way) while browsing the open issues and noticed that this one and a duplicate (#360) are still open.

georgweiss, 2021-07-26:
Thanks for pointing out, @tmijail.

closed 2022-08-10 by shroffk