ControlSystemStudio/phoebus #1677: save&restore flyway script problem

forestmojo, 2020-12-03:
hi, 

I try to run save and restore service by this command:
java -Ddbengine=mysql -Dspring.datasource.username=root -Dspring.datasource.password= -Dspring.datasource.jdbcUrl=jdbc:mysql://localhost:3306/sar -jar service-save-and-restore-4.6.6-SNAPSHOT.jar
the output like this:

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::        (v2.1.5.RELEASE)

2020-12-03 08:49:59.914 [main] INFO  org.phoebus.service.saveandrestore.application.Application - Starting Application on cow4 with PID 19668 (/home/iocusr/phoebus/services/save-and-restore/target/service-save-and-restore-4.6.6-SNAPSHOT.jar started by iocusr in /home/iocusr/phoebus/services/save-and-restore/target)
2020-12-03 08:49:59.919 [main] INFO  org.phoebus.service.saveandrestore.application.Application - No active profile set, falling back to default profiles: default
2020-12-03 08:50:00.870 [main] INFO  org.springframework.context.support.PostProcessorRegistrationDelegate$BeanPostProcessorChecker - Bean 'org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration' of type [org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration$$EnhancerBySpringCGLIB$$2a9b3d0a] is not eligible for getting processed by all BeanPostProcessors (for example: not eligible for auto-proxying)
2020-12-03 08:50:01.096 [main] INFO  org.springframework.boot.web.embedded.tomcat.TomcatWebServer - Tomcat initialized with port(s): 8080 (http)
2020-12-03 08:50:01.126 [main] INFO  org.apache.coyote.http11.Http11NioProtocol - Initializing ProtocolHandler ["http-nio-8080"]
2020-12-03 08:50:01.138 [main] INFO  org.apache.catalina.core.StandardService - Starting service [Tomcat]
2020-12-03 08:50:01.138 [main] INFO  org.apache.catalina.core.StandardEngine - Starting Servlet engine: [Apache Tomcat/9.0.19]
2020-12-03 08:50:01.193 [main] INFO  org.apache.catalina.core.ContainerBase.[Tomcat].[localhost].[/] - Initializing Spring embedded WebApplicationContext
2020-12-03 08:50:01.193 [main] INFO  org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization completed in 1225 ms
2020-12-03 08:50:01.582 [main] INFO  org.flywaydb.core.internal.license.VersionPrinter - Flyway Community Edition 5.2.4 by Boxfuse
2020-12-03 08:50:01.593 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
2020-12-03 08:50:01.714 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
2020-12-03 08:50:01.717 [main] INFO  org.flywaydb.core.internal.database.DatabaseFactory - Database: jdbc:mysql://localhost:3306/sar (MySQL 5.5)
2020-12-03 08:50:01.819 [main] INFO  org.flywaydb.core.internal.command.DbValidate - Successfully validated 28 migrations (execution time 00:00.048s)
2020-12-03 08:50:01.829 [main] INFO  org.flywaydb.core.internal.schemahistory.JdbcTableSchemaHistory - Creating Schema History table: `sar`.`flyway_schema_history`
2020-12-03 08:50:01.941 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Current version of schema `sar`: << Empty Schema >>
2020-12-03 08:50:01.942 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 1.0 - initial
2020-12-03 08:50:01.950 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Incorrect table definition; there can be only one TIMESTAMP column with CURRENT_TIMESTAMP in DEFAULT or ON UPDATE clause (SQL State: HY000 - Error Code: 1293)
2020-12-03 08:50:01.952 [main] ERROR org.flywaydb.core.internal.command.DbMigrate - Migration of schema `sar` to version 1.0 - initial failed! Please restore backups and roll back database and code!
2020-12-03 08:50:01.968 [main] WARN  org.springframework.boot.web.servlet.context.AnnotationConfigServletWebServerApplicationContext - Exception encountered during context initialization - cancelling refresh attempt: org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'snapshotController': Unsatisfied dependency expressed through field 'services'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'services': Unsatisfied dependency expressed through field 'nodeDAO'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'configDAO': Unsatisfied dependency expressed through field 'jdbcTemplate'; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'flywayInitializer' defined in class path resource [org/springframework/boot/autoconfigure/flyway/FlywayAutoConfiguration$FlywayConfiguration.class]: Invocation of init method failed; nested exception is org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException: 
Migration V1_0__initial.sql failed
----------------------------------
SQL State  : HY000
Error Code : 1293
Message    : Incorrect table definition; there can be only one TIMESTAMP column with CURRENT_TIMESTAMP in DEFAULT or ON UPDATE clause
Location   : db/migration/mysql/V1_0__initial.sql (/home/iocusr/phoebus/services/save-and-restore/target/file:/home/iocusr/phoebus/services/save-and-restore/target/service-save-and-restore-4.6.6-SNAPSHOT.jar!/BOOT-INF/classes!/db/migration/mysql/V1_0__initial.sql)
Line       : 12
Statement  : CREATE TABLE IF NOT EXISTS node(
  id INTEGER PRIMARY KEY AUTO_INCREMENT,
  name VARCHAR(50) NOT NULL,
  type VARCHAR(20) NOT NULL,
  created TIMESTAMP DEFAULT CURRENT_TIMESTAMP,
  last_modified TIMESTAMP DEFAULT CURRENT_TIMESTAMP ON UPDATE CURRENT_TIMESTAMP,
  KEY node_idx (id, name, type)
) ENGINE=InnoDB

2020-12-03 08:50:01.969 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown initiated...
2020-12-03 08:50:01.977 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Shutdown completed.
2020-12-03 08:50:01.983 [main] INFO  org.apache.catalina.core.StandardService - Stopping service [Tomcat]
2020-12-03 08:50:02.002 [main] INFO  org.springframework.boot.autoconfigure.logging.ConditionEvaluationReportLoggingListener - 

Error starting ApplicationContext. To display the conditions report re-run your application with 'debug' enabled.
2020-12-03 08:50:02.003 [main] ERROR org.springframework.boot.SpringApplication - Application run failed
org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'snapshotController': Unsatisfied dependency expressed through field 'services'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'services': Unsatisfied dependency expressed through field 'nodeDAO'; nested exception is org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'configDAO': Unsatisfied dependency expressed through field 'jdbcTemplate'; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'flywayInitializer' defined in class path resource [org/springframework/boot/autoconfigure/flyway/FlywayAutoConfiguration$FlywayConfiguration.class]: Invocation of init method failed; nested exception is org.flywaydb.core.internal.command.DbMigrate$FlywayMigrateException: 
Migration V1_0__initial.sql failed
----------------------------------
SQL State  : HY000
Error Code : 1293
Message    : Incorrect table definition; there can be only one TIMESTAMP column with CURRENT_TIMESTAMP in DEFAULT or ON UPDATE clause
Location   : db/migration/mysql/V1_0__initial.sql (/home/iocusr/phoebus/services/save-and-restore/target/file:/home/iocusr/phoebus/services/save-and-restore/target/service-save-and-restore-4.6.6-SNAPSHOT.jar!/BOOT-INF/classes!/db/migration/mysql/V1_0__initial.sql)
Line       : 12
Statement  : CREATE TABLE IF NOT EXISTS node(
  id INTEGER PRIMARY KEY AUTO_INCREMENT,
  name VARCHAR(50) NOT NULL,
  type VARCHAR(20) NOT NULL,
  created TIMESTAMP DEFAULT CURRENT_TIMESTAMP,
[... 372 lines cut ...]
        at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.populateBean(AbstractAutowireCapableBeanFactory.java:1411) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
[... cut ...]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.doResolveDependency(DefaultListableBeanFactory.java:1248) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.resolveDependency(DefaultListableBeanFactory.java:1168) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.inject(AutowiredAnnotationBeanPostProcessor.java:593) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        ... 27 more

georgweiss, 2020-12-03:
What version of Mysql are you running? 
Found this via Stackoverflow:

Changes in MySQL 5.6.5 (2012-04-10, Milestone 8)

Previously, at most one TIMESTAMP column per table could be automatically initialized or updated to the current date and time. This restriction has been lifted. Any TIMESTAMP column definition can have any combination of DEFAULT CURRENT_TIMESTAMP and ON UPDATE CURRENT_TIMESTAMP clauses. In addition, these clauses now can be used with DATETIME column definitions. For more information, see Automatic Initialization and Updating for TIMESTAMP and DATETIME.

forestmojo, 2020-12-03:
Yes, the error occurs because I use mariadb-5.5.65, when I upgrade to 10.1.48-MariaDB MariaDB Server, the new message is as following:

[iocusr@cow4]:~/phoebus/services/save-and-restore/target>java -Ddbengine=mysql -Dspring.datasource.username=root -Dspring.datasource.password= -Dspring.datasource.jdbcUrl=jdbc:mysql://localhost:3306/sar -jar service-save-and-restore-4.6.6-SNAPSHOT.jar

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::        (v2.1.5.RELEASE)

2020-12-03 15:49:09.446 [main] INFO  org.phoebus.service.saveandrestore.application.Application - Starting Application on cow4 with PID 25034 (/home/iocusr/phoebus/services/save-and-restore/target/service-save-and-restore-4.6.6-SNAPSHOT.jar started by iocusr in /home/iocusr/phoebus/services/save-and-restore/target)
2020-12-03 15:49:09.451 [main] INFO  org.phoebus.service.saveandrestore.application.Application - No active profile set, falling back to default profiles: default
2020-12-03 15:49:10.378 [main] INFO  org.springframework.context.support.PostProcessorRegistrationDelegate$BeanPostProcessorChecker - Bean 'org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration' of type [org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration$$EnhancerBySpringCGLIB$$d88f362a] is not eligible for getting processed by all BeanPostProcessors (for example: not eligible for auto-proxying)
2020-12-03 15:49:10.590 [main] INFO  org.springframework.boot.web.embedded.tomcat.TomcatWebServer - Tomcat initialized with port(s): 8080 (http)
2020-12-03 15:49:10.610 [main] INFO  org.apache.coyote.http11.Http11NioProtocol - Initializing ProtocolHandler ["http-nio-8080"]
2020-12-03 15:49:10.621 [main] INFO  org.apache.catalina.core.StandardService - Starting service [Tomcat]
2020-12-03 15:49:10.622 [main] INFO  org.apache.catalina.core.StandardEngine - Starting Servlet engine: [Apache Tomcat/9.0.19]
2020-12-03 15:49:10.679 [main] INFO  org.apache.catalina.core.ContainerBase.[Tomcat].[localhost].[/] - Initializing Spring embedded WebApplicationContext
2020-12-03 15:49:10.679 [main] INFO  org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization completed in 1181 ms
2020-12-03 15:49:11.077 [main] INFO  org.flywaydb.core.internal.license.VersionPrinter - Flyway Community Edition 5.2.4 by Boxfuse
2020-12-03 15:49:11.087 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
2020-12-03 15:49:11.207 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
2020-12-03 15:49:11.210 [main] INFO  org.flywaydb.core.internal.database.DatabaseFactory - Database: jdbc:mysql://localhost:3306/sar (MySQL 5.5)
2020-12-03 15:49:11.326 [main] INFO  org.flywaydb.core.internal.command.DbValidate - Successfully validated 28 migrations (execution time 00:00.043s)
2020-12-03 15:49:11.337 [main] INFO  org.flywaydb.core.internal.schemahistory.JdbcTableSchemaHistory - Creating Schema History table: `sar`.`flyway_schema_history`
2020-12-03 15:49:11.489 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Current version of schema `sar`: << Empty Schema >>
2020-12-03 15:49:11.490 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 1.0 - initial
2020-12-03 15:49:12.240 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 2.0 - add provider
2020-12-03 15:49:12.432 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 3.0 - add snapshot name
2020-12-03 15:49:12.582 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 4.0 - rename root node
2020-12-03 15:49:12.606 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 5.0 - alter snapshot pv simplify config pv
2020-12-03 15:49:12.743 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
2020-12-03 15:49:12.871 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
2020-12-03 15:49:12.988 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
2020-12-03 15:49:13.139 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
2020-12-03 15:49:13.674 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
2020-12-03 15:49:13.816 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
2020-12-03 15:49:13.967 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
[... cut ...]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.doResolveDependency(DefaultListableBeanFactory.java:1248) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.support.DefaultListableBeanFactory.resolveDependency(DefaultListableBeanFactory.java:1168) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        at org.springframework.beans.factory.annotation.AutowiredAnnotationBeanPostProcessor$AutowiredFieldElement.inject(AutowiredAnnotationBeanPostProcessor.java:593) ~[spring-beans-5.1.7.RELEASE.jar!/:5.1.7.RELEASE]
        ... 27 more

georgweiss, 2020-12-03:
I know I need to update the README, but MariaDB support was added after Mysql and Postgresql.
Please use  -Ddbengine=mariadb

forestmojo, 2020-12-03:
hi, Georg:

Thanks a lot for your reply, but as your suggestion, I try -Ddbengine=mariadb, still errors. Maybe I should change back to mysql on centos later?

[iocusr@cow4]:~/phoebus/services/save-and-restore/target>mysql -u root
Welcome to the MariaDB monitor.  Commands end with ; or \g.
Your MariaDB connection id is 14
Server version: 10.1.48-MariaDB MariaDB Server

Copyright (c) 2000, 2018, Oracle, MariaDB Corporation Ab and others.

Type 'help;' or '\h' for help. Type '\c' to clear the current input statement.

MariaDB [(none)]> drop database sar;
Query OK, 10 rows affected (0.24 sec)

MariaDB [(none)]> create database sar;    
Query OK, 1 row affected (0.00 sec)

MariaDB [(none)]> Bye
[iocusr@cow4]:~/phoebus/services/save-and-restore/target>java -Ddbengine=mariadb -Dspring.datasource.username=root -Dspring.datasource.password= -Dspring.datasource.jdbcUrl=jdbc:mysql://localhost:3306/sar -jar service-save-and-restore-4.6.6-SNAPSHOT.jar     

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::        (v2.1.5.RELEASE)

2020-12-03 22:26:57.480 [main] INFO  org.phoebus.service.saveandrestore.application.Application - Starting Application on cow4 with PID 29968 (/home/iocusr/phoebus/services/save-and-restore/target/service-save-and-restore-4.6.6-SNAPSHOT.jar started by iocusr in /home/iocusr/phoebus/services/save-and-restore/target)
2020-12-03 22:26:57.488 [main] INFO  org.phoebus.service.saveandrestore.application.Application - No active profile set, falling back to default profiles: default
2020-12-03 22:26:58.425 [main] INFO  org.springframework.context.support.PostProcessorRegistrationDelegate$BeanPostProcessorChecker - Bean 'org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration' of type [org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration$$EnhancerBySpringCGLIB$$a3bd2265] is not eligible for getting processed by all BeanPostProcessors (for example: not eligible for auto-proxying)
2020-12-03 22:26:58.630 [main] INFO  org.springframework.boot.web.embedded.tomcat.TomcatWebServer - Tomcat initialized with port(s): 8080 (http)
2020-12-03 22:26:58.649 [main] INFO  org.apache.coyote.http11.Http11NioProtocol - Initializing ProtocolHandler ["http-nio-8080"]
2020-12-03 22:26:58.660 [main] INFO  org.apache.catalina.core.StandardService - Starting service [Tomcat]
2020-12-03 22:26:58.661 [main] INFO  org.apache.catalina.core.StandardEngine - Starting Servlet engine: [Apache Tomcat/9.0.19]
2020-12-03 22:26:58.718 [main] INFO  org.apache.catalina.core.ContainerBase.[Tomcat].[localhost].[/] - Initializing Spring embedded WebApplicationContext
2020-12-03 22:26:58.719 [main] INFO  org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization completed in 1179 ms
2020-12-03 22:26:59.105 [main] INFO  org.flywaydb.core.internal.license.VersionPrinter - Flyway Community Edition 5.2.4 by Boxfuse
2020-12-03 22:26:59.115 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
2020-12-03 22:26:59.175 [main] INFO  com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
2020-12-03 22:26:59.180 [main] INFO  org.flywaydb.core.internal.database.DatabaseFactory - Database: jdbc:mysql://localhost:3306/sar (MySQL 10.1)
2020-12-03 22:26:59.257 [main] INFO  org.flywaydb.core.internal.command.DbValidate - Successfully validated 28 migrations (execution time 00:00.037s)
2020-12-03 22:26:59.263 [main] INFO  org.flywaydb.core.internal.schemahistory.JdbcTableSchemaHistory - Creating Schema History table: `sar`.`flyway_schema_history`
2020-12-03 22:26:59.431 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Current version of schema `sar`: << Empty Schema >>
2020-12-03 22:26:59.432 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 1.0 - initial
2020-12-03 22:27:00.364 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 2.0 - add provider
2020-12-03 22:27:00.522 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 3.0 - add snapshot name
2020-12-03 22:27:00.739 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 4.0 - rename root node
2020-12-03 22:27:00.763 [main] INFO  org.flywaydb.core.internal.command.DbMigrate - Migrating schema `sar` to version 5.0 - alter snapshot pv simplify config pv
2020-12-03 22:27:00.909 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
2020-12-03 22:27:01.021 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
2020-12-03 22:27:01.130 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
2020-12-03 22:27:01.280 [main] WARN  org.flywaydb.core.internal.sqlscript.DefaultSqlScriptExecutor - DB: Table 'mysql.column_stats' doesn't exist (SQL State: 42S02 - Error Code: 1146)
[... cut ...]
2020-12-03 22:27:08.581 [main] INFO  org.apache.catalina.core.StandardService - Stopping service [Tomcat]
2020-12-03 22:27:08.595 [main] INFO  org.springframework.boot.autoconfigure.logging.ConditionEvaluationReportLoggingListener - 

Error starting ApplicationContext. To display the conditions report re-run your application with 'debug' enabled.
2020-12-03 22:27:08.596 [main] ERROR org.springframework.boot.SpringApplication - Application run failed
  

Best wishes
Geyang 2020-12-03

###############

georgweiss, 2020-12-03:
Need to ask @geniejhang for assistance as I have no MariaDB environment.

geniejhang, 2020-12-03:
I see @forestmojo JDBC driver is set with mysql. Though I tested MariaDB working with that option, I recommend you change that to mariadb: -Dspring.datasource.jdbcUrl=jdbc:mariadb://localhost:3306/sar
We're using MariaDB 10.4.12 version and it's running fine with it.
It looks like you're using 10.1.48, which I never tested. I tested MariaDB 10.4.15 and it didn't work due to flyway support issue as of 2 weeks ago.

geniejhang, 2020-12-03:
Confirmed current save&restore version works with 10.4.17 as well.

forestmojo, 2020-12-03:
Thanks a lot for your information. I tried mysql 5.7.32, it works fine too.

forestmojo, 2020-12-03:
but how to invoke GUI  as following link? 
https://control-system-studio.readthedocs.io/en/latest/_images/screenshot1.png
I can't find it in the phoebus->menu->application?
I have add "org.phoebus.applications.saveandrestore/jmasar.service.url=http://localhost:8080" to settings.ini.

> 
> 
> Confirmed current save&restore version works with 10.4.17 as well.

geniejhang, 2020-12-03:
What you've run is just backend part. You'll need Phoebus client to interact with GUI.

forestmojo, 2020-12-03:
yes, but how to run phoebus client?  Phoebus.sh -app save-and restore???

georgweiss, 2020-12-03:
Launch from the application menu:
[image]

kasemir, 2020-12-03:
Sounds like user has a product that doesn't include the save/restore UI.
https://github.com/ControlSystemStudio/phoebus/tree/master/phoebus-product lists some site-specific products.
Maybe add link to ESS, a product which I assume includes save/restore?

forestmojo, 2020-12-04:
hi, Kay:

thank you very much. I use the linux binary version downloaded from SNS. seems it has no save/restore GUI.
I tried https://github.com/shroffk/nsls2-phoebus, it can work now.
But when I follow the steps to run build.sh, at the end, the result is as follows, would you please give me some suggestions?

[INFO] app-alarm-datasouce 4.6.5-SNAPSHOT ................. SUCCESS [  0.141 s]
[INFO] app-scan 4.6.5-SNAPSHOT ............................ SUCCESS [  0.007 s]
[INFO] app-scan-model 4.6.5-SNAPSHOT ...................... SUCCESS [  0.317 s]
[INFO] app-scan-client 4.6.5-SNAPSHOT ..................... SUCCESS [  0.124 s]
[INFO] app-scan-ui 4.6.5-SNAPSHOT ......................... SUCCESS [  0.381 s]
[INFO] app-channel 4.6.5-SNAPSHOT ......................... SUCCESS [  0.007 s]
[INFO] app-channel-channelfinder 4.6.5-SNAPSHOT ........... SUCCESS [  0.185 s]
[INFO] app-channel-utility 4.6.5-SNAPSHOT ................. SUCCESS [  0.087 s]
[INFO] app-channel-views 4.6.5-SNAPSHOT ................... SUCCESS [  0.270 s]
[INFO] app-perfmon 4.6.5-SNAPSHOT ......................... SUCCESS [  0.077 s]
[INFO] app-pace 4.6.5-SNAPSHOT ............................ SUCCESS [  0.208 s]
[INFO] app-save-and-restore 4.6.5-SNAPSHOT ................ SUCCESS [  0.008 s]
[INFO] app-save-and-restore-model 4.6.5-SNAPSHOT .......... SUCCESS [  2.194 s]
[INFO] app-save-and-restore-ui 4.6.5-SNAPSHOT ............. SUCCESS [  1.146 s]
[INFO] app-console 4.6.5-SNAPSHOT ......................... SUCCESS [  0.140 s]
[INFO] app-trends 4.6.5-SNAPSHOT .......................... SUCCESS [  0.007 s]
[INFO] app-trends-rich-context-menu 4.6.5-SNAPSHOT ........ SUCCESS [  0.095 s]
[INFO] app-trends-simple-context-menu 4.6.5-SNAPSHOT ...... SUCCESS [  0.107 s]
[INFO] product 4.6.5-SNAPSHOT ............................. SUCCESS [ 13.675 s]
[INFO] services 4.6.5-SNAPSHOT ............................ SUCCESS [  0.008 s]
[INFO] service-alarm-server 4.6.5-SNAPSHOT ................ SUCCESS [  1.580 s]
[INFO] service-archive-engine 4.6.5-SNAPSHOT .............. SUCCESS [  0.381 s]
[INFO] service-scan-server 4.6.5-SNAPSHOT ................. SUCCESS [  0.525 s]
[INFO] service-alarm-logger 4.6.5-SNAPSHOT ................ SUCCESS [  5.625 s]
[INFO] service-alarm-config-logger 4.6.5-SNAPSHOT ......... SUCCESS [  3.140 s]
[INFO] service-save-and-restore 4.6.6-SNAPSHOT ............ SUCCESS [  3.474 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  53.079 s
[INFO] Finished at: 2020-12-04T08:58:36+08:00
[INFO] ------------------------------------------------------------------------
[INFO] Scanning for projects...
[ERROR] [ERROR] Some problems were encountered while processing the POMs:
[FATAL] Non-resolvable parent POM for org.phoebus:nsls2-products:[unknown-version]: Could not find artifact org.phoebus:parent:pom:4.6.2-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 7, column 11
 @ 
[ERROR] The build could not read 1 project -> [Help 1]
[ERROR]   
[ERROR]   The project org.phoebus:nsls2-products:[unknown-version] (/home/iocusr/github/nsls2-phoebus/products/pom.xml) has 1 error
[ERROR]     Non-resolvable parent POM for org.phoebus:nsls2-products:[unknown-version]: Could not find artifact org.phoebus:parent:pom:4.6.2-SNAPSHOT and 'parent.relativePath' points at wrong local POM @ line 7, column 11 -> [Help 2]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/ProjectBuildingException
[ERROR] [Help 2] http://cwiki.apache.org/confluence/display/MAVEN/UnresolvableModelException

georgweiss, 2020-12-04:
Cloning the project and following the instructions for a Maven build (as described in the README) should give you the full feature set, including save & restore. This is basically what we do at ESS, we only add customizations related to color and font definitions.

If you just want to try the ESS build, feel free to browse https://artifactory.esss.lu.se/ui/packages/gav:%2F%2Fse.europeanspallationsource.ics:ess-cs-studio-phoebus?name=ess-cs-studio-phoebus&type=packages where we have packages for recent versions (please avoid 4.6.4, as this has a bug related to archiver settings) and all supported platforms. Note that the packages come without a settings file, so default settings will be used.  You will need to put together a settings file to make an application like save & restore work correctly.

georgweiss, 2020-12-07:
@forestmojo , have you been able to resolve your issues?

forestmojo, 2020-12-07:
> 
> 
> @forestmojo , have you been able to resolve your issues?

hi, geor:

almost solve my problem.  by this topic, I know more about phoebus. thanks a lot for your help.

closed 2020-12-07 by forestmojo