ControlSystemStudio/phoebus #1693: No logbook entries from data browser

kasemir, 2020-12-14:
The DatabrowserAdapterFactory in app-trends-simple-adapters supports both emails and logbook entries from the data browser context menu.

The DatabrowserAdapterFactory in app-trends-rich-adapters fundamentally implements that as well:
```
public <T> Optional<T> adapt(Object adaptableObject, Class<T> adapterType)
 ...
    if (adapterType.isAssignableFrom(EmailEntry.class))
 ...
    else if (adapterType.isAssignableFrom(LogEntry.class))
```

But since it doesn't declare support for LogEntry, only Email is supported:

```
    private static final List<? extends Class> adaptableTypes = Arrays.asList(EmailEntry.class);
```

shroffk, 2020-12-14:
@kasemir @georgweiss I am sorry I missed this

shroffk, 2020-12-14:
I switched the implementation from simple to rich and missed this while making the switch

kasemir, 2020-12-14:
Wasn't sure if it's a typo or has deeper meaning, like showing how to select what to support..
Looks good with the PR. I think we talked about having 'rich' and 'simple'. If 'rich' is better, I'm OK with keeping only that, but I think there the idea was to demonstrate how to do things.

shroffk, 2020-12-14:
> If 'rich' is better, I'm OK with keeping only that, but I think there the idea was to demonstrate how to do things.
+1
While we all can surely agree on moving the rich adapter... This is a decent example which is simple to understand.

closed 2020-12-14 by shroffk