ControlSystemStudio/phoebus #1697: Create log from Databrowser throws NPE

georgweiss, 2020-12-15:
java.lang.NullPointerException
	at org.phoebus.logbook.ui.write.LogEntryModel.<init>(LogEntryModel.java:165)

Root cause is that a plt file has no content type string set. 
I can fix, but need agreement on content type, e.g. "plt".

kasemir, 2020-12-15:
`org.phoebus.logbook.Attachment` has no javadoc to describe what the content type means: May it be null? Are they supposed to be MIME types? The only use I see is `if (...toLowerCase().startsWith("image"))`, so should be safe to have "plt", "bob", "opi", "pvs", ... for other files used by CS-Studio.

georgweiss, 2020-12-16:
I guess plt would be fine, even though I would prefer using MIME types. Since plt is xml I think it would be more along common practices to define something like text/xml;csstype=plt, text/xml;csstype=bob etc.

closed 2020-12-19 by georgweiss